KLayout/klayout #746: why klayout python package miss lay module

fzr4j, 2021-03-11:
l download klayout package from pypi.org . I can find lay module on klayout manual , but can not find at this package.
[image]

klayoutmatthias, 2021-03-13:
Hi,

There is no lay module in the PyPI package. This implies a dependency to Qt which raises deployment and legal questions, so this has been dropped from the PyPI versions. The lay module is available in the Python module packaged with the Linux installers.

Matthias

klayoutmatthias, 2021-04-14:
Can I close this ticket?

fzr4j, 2021-04-24:
> Can I close this ticket?

yes, thank you for reply

fzr4j, 2021-04-24:
> Can I close this ticket?
yes , thank you for your replay

closed 2021-04-24 by klayoutmatthias