KLayout/klayout #750: PCB gerber import: Incorrect handling of D02 operations

neinseg, 2021-03-19:
When importing a PCB gerber file through File -> Import -> Gerber PCB -> New Project, KLayout incorrectly interprets D02 operations inside G36/G37 regions. ```test1.gbr``` below (download: [testfiles.zip](https://github.com/KLayout/klayout/files/6173438/testfiles.zip)) should render a rectangle. Ucamco's reference gerber viewer agrees.

KLayout v0.26.11 instead renders a triangle. The missing point is the one described by the D02 "move" operation. When duplicating the coordinates of the D02 operation with an additional D01 operation, KLayout again produces correct output. See ```test2.gbr```. A screenshot of KLayout with both files loaded is embedded below. Thanks to @mx-yah for reporting this on the gerbolyze tracker, https://github.com/jaseg/gerbolyze/issues/12

The issue seems to stem from KLayout completely ignoring the coordinate when handling D02 statements:
https://github.com/KLayout/klayout/blob/7d4310d343a53f2682ec5daac50a7f73ab9db4d0/src/plugins/streamers/pcb/db_plugin/dbRS274XReader.cc#L510-L526

```test1.gbr```:
```
%FSLAX46Y46*%
%MOMM*%
%LPD*%
G01*
%ADD10C,0.050000*%
D10*
%LPD*%
G36*
X0070000000Y0050000000D02*
X0020000000Y0050000000D01*
X0020000000Y0080000000D01*
X0070000000Y0080000000D01*
G37*
M02*
```

```test2.gbr```:
```
%FSLAX46Y46*%
%MOMM*%
%LPD*%
G01*
%ADD10C,0.050000*%
D10*
%LPD*%
G36*
X0070000000Y0050000000D02*
X0070000000Y0050000000D01*
X0020000000Y0050000000D01*
X0020000000Y0080000000D01*
X0070000000Y0080000000D01*
G37*
M02*
```

KLayout with both files loaded:
[image]

Here's ```test1.gbr``` in [Ucamco's reference gerber viewer](https://gerber-viewer.ucamco.com/) for comparison:
[image]

klayoutmatthias, 2021-03-21:
@neinseg Thanks a lot for this nice test case. I'll take a look.

Kind regards,

Matthias

klayoutmatthias, 2021-04-14:
As 0.27 is ahead I don't think I'll port the fix to 0.26.12. Let me know if you need to and reopen the bug in this case.

closed 2021-04-14 by klayoutmatthias