KLayout/klayout #758: possible problem when reading DXF files

reinaldgerhardt2, 2021-03-25:
I am using Klayout ver. 0.26.11
Attached are two dxf files:
test1.dxf contains only one "lwpolyline" entity (a closed square at coordinates (0,0)). Klayout correctly displays the entity as a closed polygon.
test2.dxf contains the same lwpolyline entity as test1 plus an entity of type "solid" next to it (a triangle). Klayout correctly displays the triangle as a closed polygon, but the square now gets displayed as a path, which is not what want.
Is this a bug or a feature?
Thanks, Reinald
[dxf.zip](https://github.com/KLayout/klayout/files/6207397/dxf.zip)

klayoutmatthias, 2021-04-01:
@reinaldgerhardt2 

Thanks for the test cases and this report. 

You will get the desired results if you chose a suitable mode in the "LWPOLYLINE" reader mode. Goto File/Reader Options and on the DXF page select this mode:

[image]

Background is: DXF has no concept of "filled area". The default "automatic" mode will interpret the presence of SOLID that in your flavour of drawing you plan to use SOLID to represent filled area. HATCH will trigger the same decision. In both cases, POLYLINE and LWPOLYLINEs will be taken as "hollow". Using the right mode will make KLayout interpret them as "filled".

I can only emphasize again that DXF should not be used for representing physical mask data. It's not a well-defined format like GDS (even considering of the inherent weaknesses of GDS). The risk of loss of information or pattern corruption is too high.

Matthias

reinaldgerhardt2, 2021-04-16:
> @reinaldgerhardt2 Was that reply useful? I'd close this ticket as it's not something I can change.

Hi Mathias,Thank you so much for your reply (I missed your first email thanks to all the spam that's filling up my inbox). That was extremely helpful indeed. I would not have looked for that "Reader Options" Menu on my own. Problem solved!At my company we are using standard CAD software (like Autocad) to generate the basic design for a single chip. I am using Klayout to convert the DXF output to GDS in a first step. In a second step, I am reading said GDS file into gdspy and automatically create a photomask which has some 50,000 instances of said design. I suppose I could do all of that from within Klayout, but I haven't had the time yet to look into it.Thank you so much for providing this great tool for free. I have used a commercial software (don't quite remember the name) a long time ago, but I like Klayout so much better. The speed of the display never stops amazing me. Excellent work!Viele Gruesse aus Michigan,Reinald

klayoutmatthias, 2021-04-18:
@reinaldgerhardt2 Thanks, that was good feedback :) 

I'll close the ticket.

closed 2021-04-18 by klayoutmatthias