KLayout/klayout #765: Feature request: enable editing of salt packages in the macro editor

mustafacc, 2021-04-04:
Hello,

Currently, packages downloaded from the salt package manger are read-only in the macro editor. 

It would be convenient if there was a checkbox (ticked on by default) that will disable the read-only functionality and allow us to edit the packages directly from the macro-editor instead of editing the files manually by navigating to the salt directory.

Thanks!

klayoutmatthias, 2021-04-06:
@mustafacc The idea is not to let someone edit the packages. Otherwise you'd create a support hell for someone who is trying to help with a problem in such packages.

edaljohani, 2022-07-22:
@klayoutmatthias is there a solution for developers? Currently, the IDE assumes you've downloaded a package, not developing one, so it's read-only. The workaround is using some other text editor but wondering if there's a built-in solution I'm missing.

Secondly, editing a package also seems to require restarting klayout or the changes won't take effect when loading the libraries in, say the pymacros tab. Thanks!

klayoutmatthias, 2022-07-22:
TL;DR Use any other place to develop packages and use KLAYOUT_PATH to point KLayout there.

In length: there is.

Basically you do not need to develop packages in "salt". This folder is reserved for packages managed by KLayout's package manager. You can develop packages in any other folder using the same structure than a salt package.

For example, you develop your package in ```~/mypackage``` with folders "macros" for generic Ruby macros, "ruby" for ruby modules, "pymacros" for generic python macros, "python" for python module, "tech" for technology setups (with tech specific folders) etc. This is the same structure you use when preparing salt package. For package deployment you simply add a grain.xml file.

So now, if you want to test your package, run KLayout with

```
klayout -j ~/mypackage
```

or

```
KLAYOUT_PATH=~/mypackage klayout
```

(similar on Windows, substitute windows paths).

Eventually salt packages are not different from folders specified in ```$KLAYOUT_PATH```, except that salt packages are implicitly looked up and they are treated read-only.

Regarding your problem with editing files and not seeing the updates: I don't know your specific setup, but KLayout does not automatically run files once you edited them. You need to explicitly run them. This can be tricky if you develop Python modules for example. If you use "import", the files are loaded exactly once. It may be possible to fool the Python module manager to reload them again on "import", but the easier way is probably to execute the file you have edited. If you want to be sure, I'm afraid the best way is to restart KLayout.

In general I'd recommend a structured approach, maybe not even involving KLayout initially. I do not know the things your are developing. If for example you develop Python PCells, you can do so inside a Python module and develop that latter with the klayout Python module from PyPI alone. This can be done conveniently with the IDE of your choice (e.g. PyCharm). You can also add unit tests which I strongly recommend. So once you finished that, it is easy to wrap the whole thing into a generic KLayout macro, add a grain.xml and deploy that whole package to Salt.

Although I'm not a PyCharm user myself, I got notice from users who successfully registered KLayout as a Python interpreter inside PyCharm. Using that setup, you should be able to develop even KLayout UI applications.

The internal IDE is not a match to heavy IDEs such as PyCharm, so I'd recommend it only for debugging and lean development of small features.

Regards,

Matthias

mustafacc, 2022-07-23:
Thank you Matthias. I'm very curious your comment regarding registering KLayout's Python interpreter in PyCharm. How does PyCharm interpret debugging and application callbacks?

Would you be able to connect me to the users who managed to get this to work? I'd be happy to write up an application note to summarize my learnings.

Cheers,
Mustafa

klayoutmatthias, 2022-07-23:
You can check the forum: https://www.klayout.de/forum. I couple of entries refer to using PyCharm, so I assume there is some experience, but not from my side.

With "klayout -b -r your_python_script.py", KLayout basically acts as a Python interpreter. One suggestion was to wrap that in a script and use that for the Python interpreter in PyCharm. I don't know however, if and how debugging works in that case.

Matthias