KLayout/klayout #766: Right Click Cell Backtrace

yrrapt, 2021-04-05:
When I right click on a cell on the cell list on the left I receive a back trace dialog with the information below.

This happens regularly but not always.

Let me know if there is any more information you'd like me to provide to to assist in tracking down the cause.

```
Signal number: 11
Address: 0x3a7000007ea
Program Version: KLayout 0.27 (2021-03-30 r729975ba)

Backtrace:
libklayout_lay.so.0 +0x2a769f lay::enable_signal_handler_gui(bool) [??:?]
/lib/x86_64-linux-gnu/libpthread.so.0 +0x12730 __restore_rt [??:?]
libklayout_laybasic.so.0 +0x2efb00 void std::vector<lay::CellView, std::allocator<lay::CellView> >::_M_realloc_insert<lay::CellView const&>(__gnu_cxx::__normal_iterator<lay::CellView*, std::vector<lay::CellView, std::allocator<lay::CellView> > >, lay::CellView const&) [??:?]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x75caeb QTreeView::isIndexHidden(QModelIndex const&) const [qabstractitemmodel.h:393]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x75b33f QTreeView::visualRect(QModelIndex const&) const [qtreeview.cpp:1045]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x71acaa QAbstractItemView::update(QModelIndex const&) [qabstractitemview.cpp:3166]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x71ffd3 QAbstractItemView::currentChanged(QModelIndex const&, QModelIndex const&) [qabstractitemmodel.h:66]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x7654b8 QTreeView::currentChanged(QModelIndex const&, QModelIndex const&) [qtreeview.cpp:3725]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x1acc90 QMetaObject::activate(QObject*, QMetaObject const*, int, void**) [qobject.cpp:3580]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x76d86a QItemSelectionModel::currentChanged(QModelIndex const&, QModelIndex const&) [moc_qitemselectionmodel.cpp:173]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x76d9db QItemSelectionModel::setCurrentIndex(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) [qitemselectionmodel.cpp:1176]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x721a6f operator-(QPoint const&, QPoint const&) [qpoint.h:171 (discriminator 4)]
libklayout_laybasic.so.0 +0x35fbd7 void db::polygon_contour<int>::assign<__gnu_cxx::__normal_iterator<db::point<int>*, std::vector<db::point<int>, std::allocator<db::point<int> > > >, db::fixpoint_trans<int> >(__gnu_cxx::__normal_iterator<db::point<int>*, std::vector<db::point<int>, std::allocator<db::point<int> > > >, __gnu_cxx::__normal_iterator<db::point<int>*, std::vector<db::point<int>, std::allocator<db::point<int> > > >, db::fixpoint_trans<int>, bool, bool, bool, bool) [??:?]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x28d2ef QWidget::event(QEvent*) [qwidget.cpp:8861]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x613f4e QFrame::event(QEvent*) [qframe.cpp:559]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x725e83 QAbstractItemView::viewportEvent(QEvent*) [qabstractitemview.cpp:1644]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x7643eb QTreeView::viewportEvent(QEvent*) [qtreeview.cpp:1253]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x198c3c QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) [qcoreapplication.cpp:1065]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x23aa2c QApplicationPrivate::notify_helper(QObject*, QEvent*) [qapplication.cpp:4566]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x24140c QApplication::notify(QObject*, QEvent*) [qapplication.cpp:4114 (discriminator 4)]
libklayout_lay.so.0 +0x12ee46 lay::GuiApplication::notify(QObject*, QEvent*) [??:?]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x198afb QCoreApplication::notifyInternal(QObject*, QEvent*) [qcoreapplication.cpp:968]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x240b4b QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) [qcoreapplication.h:231]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x2b3d94 QETWidget::translateMouseEvent(_XEvent const*) [qapplication_x11.cpp:4558]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x2b29e0 QApplication::x11ProcessEvent(_XEvent*) [qapplication_x11.cpp:3953]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x2da07c x11EventSourceDispatch [qguieventdispatcher_glib.cpp:146]
/lib/x86_64-linux-gnu/libglib-2.0.so.0 +0x4df2e g_main_dispatch [gmain.c:3186]
/lib/x86_64-linux-gnu/libglib-2.0.so.0 +0x4e1c8 g_main_context_iterate [gmain.c:3920]
/lib/x86_64-linux-gnu/libglib-2.0.so.0 +0x4e25c g_main_context_iteration [gmain.c:3982]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x1c58e1 QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) [qeventdispatcher_glib.cpp:425]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x2da207 QPaintDevice::devType() const [qglobal.h:2441]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x19738f QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) [qeventloop.cpp:150]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x19765e QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) [qeventloop.cpp:203]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x19cb2a QCoreApplication::exec() [qcoreapplication.cpp:1229]
libklayout_lay.so.0 +0x130228 lay::GuiApplication::exec() [??:?]
libklayout_lay.so.0 +0x13102c lay::ApplicationBase::run() [??:?]
./klayout +0x3fc8 ?? [??:0]
libklayout_rba.so.0 +0x226b0 rba::RubyInterpreter::set_debugger_scope(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) [??:?]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x1b47a1 rb_iter_break_value [??:?]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x1c238b rb_yield_block [??:?]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x1ba306 rb_check_funcall [??:?]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x1bdef4 rb_check_funcall [??:?]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x94f14 rb_check_copyable [??:?]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x96dbd ruby_exec_node [??:?]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x9926e ruby_run_node [??:?]
libklayout_rba.so.0 +0x22b70 rba::RubyInterpreter::initialize(int&, char**, int (*)(int&, char**)) [??:?]
./klayout +0x38a7 ?? [??:0]
/lib/x86_64-linux-gnu/libc.so.6 +0x2409b __libc_start_main [libc-start.c:342]
./klayout +0x39ea ?? [??:0]
```

klayoutmatthias, 2021-04-06:
@yrrapt Thanks for this backtrace. It's already helpful, but debugging that is going to be some pain ...

klayoutmatthias, 2021-04-07:
@yrrapt The back trace is already good, but I could not really reproduce the problem. Do you have any hints for me? Like maybe explaining what you did before that? Have you been editing? Using PCells?

Thanks,

Matthias

yrrapt, 2021-04-11:
@klayoutmatthias 
Apologies for the delay I wanted to replicate the issue exactly and didn't get a chance until today.
The exact where this just happened to me is this:

I was editing the `dac_bias` of the attached gds (saved immediately after the error popped up).
I tried to right click on `dac_unit_cell` and the following back trace popped up:
```
[amsat_txrx_ic.gds.zip](https://github.com/KLayout/klayout/files/6292169/amsat_txrx_ic.gds.zip)

Signal number: 11
Address: 0x3a7000007ea
Program Version: KLayout 0.27 (2021-03-30 r729975ba)

Backtrace:
libklayout_lay.so.0 +0x2a769f lay::enable_signal_handler_gui(bool) [??:?]
/lib/x86_64-linux-gnu/libpthread.so.0 +0x12730 __restore_rt [??:?]
libklayout_laybasic.so.0 +0x2efb00 void std::vector<lay::CellView, std::allocator<lay::CellView> >::_M_realloc_insert<lay::CellView const&>(__gnu_cxx::__normal_iterator<lay::CellView*, std::vector<lay::CellView, std::allocator<lay::CellView> > >, lay::CellView const&) [??:?]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x75caeb QTreeView::isIndexHidden(QModelIndex const&) const [qabstractitemmodel.h:393]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x75b33f QTreeView::visualRect(QModelIndex const&) const [qtreeview.cpp:1045]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x71acaa QAbstractItemView::update(QModelIndex const&) [qabstractitemview.cpp:3166]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x71ffd3 QAbstractItemView::currentChanged(QModelIndex const&, QModelIndex const&) [qabstractitemmodel.h:66]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x7654b8 QTreeView::currentChanged(QModelIndex const&, QModelIndex const&) [qtreeview.cpp:3725]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x1acc90 QMetaObject::activate(QObject*, QMetaObject const*, int, void**) [qobject.cpp:3580]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x76d86a QItemSelectionModel::currentChanged(QModelIndex const&, QModelIndex const&) [moc_qitemselectionmodel.cpp:173]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x76d9db QItemSelectionModel::setCurrentIndex(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) [qitemselectionmodel.cpp:1176]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x721a6f operator-(QPoint const&, QPoint const&) [qpoint.h:171 (discriminator 4)]
libklayout_laybasic.so.0 +0x35fbd7 void db::polygon_contour<int>::assign<__gnu_cxx::__normal_iterator<db::point<int>*, std::vector<db::point<int>, std::allocator<db::point<int> > > >, db::fixpoint_trans<int> >(__gnu_cxx::__normal_iterator<db::point<int>*, std::vector<db::point<int>, std::allocator<db::point<int> > > >, __gnu_cxx::__normal_iterator<db::point<int>*, std::vector<db::point<int>, std::allocator<db::point<int> > > >, db::fixpoint_trans<int>, bool, bool, bool, bool) [??:?]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x28d2ef QWidget::event(QEvent*) [qwidget.cpp:8861]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x613f4e QFrame::event(QEvent*) [qframe.cpp:559]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x725e83 QAbstractItemView::viewportEvent(QEvent*) [qabstractitemview.cpp:1644]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x7643eb QTreeView::viewportEvent(QEvent*) [qtreeview.cpp:1253]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x198c3c QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) [qcoreapplication.cpp:1065]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x23aa2c QApplicationPrivate::notify_helper(QObject*, QEvent*) [qapplication.cpp:4566]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x24140c QApplication::notify(QObject*, QEvent*) [qapplication.cpp:4114 (discriminator 4)]
libklayout_lay.so.0 +0x12ee46 lay::GuiApplication::notify(QObject*, QEvent*) [??:?]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x198afb QCoreApplication::notifyInternal(QObject*, QEvent*) [qcoreapplication.cpp:968]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x240b4b QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) [qcoreapplication.h:231]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x2b3d94 QETWidget::translateMouseEvent(_XEvent const*) [qapplication_x11.cpp:4558]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x2b29e0 QApplication::x11ProcessEvent(_XEvent*) [qapplication_x11.cpp:3953]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x2da07c x11EventSourceDispatch [qguieventdispatcher_glib.cpp:146]
/lib/x86_64-linux-gnu/libglib-2.0.so.0 +0x4df2e g_main_dispatch [gmain.c:3186]
/lib/x86_64-linux-gnu/libglib-2.0.so.0 +0x4e1c8 g_main_context_iterate [gmain.c:3920]
/lib/x86_64-linux-gnu/libglib-2.0.so.0 +0x4e25c g_main_context_iteration [gmain.c:3982]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x1c58e1 QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) [qeventdispatcher_glib.cpp:425]
/lib/x86_64-linux-gnu/libQtGui.so.4 +0x2da207 QPaintDevice::devType() const [qglobal.h:2441]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x19738f QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) [qeventloop.cpp:150]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x19765e QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) [qeventloop.cpp:203]
/lib/x86_64-linux-gnu/libQtCore.so.4 +0x19cb2a QCoreApplication::exec() [qcoreapplication.cpp:1229]
libklayout_lay.so.0 +0x130228 lay::GuiApplication::exec() [??:?]
libklayout_lay.so.0 +0x13102c lay::ApplicationBase::run() [??:?]
./klayout +0x3fc8 ?? [??:0]
libklayout_rba.so.0 +0x226b0 rba::RubyInterpreter::set_debugger_scope(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) [??:?]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x1b47a1 rb_iter_break_value [??:?]
[... cut ...]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x94f14 rb_check_copyable [??:?]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x96dbd ruby_exec_node [??:?]
/lib/x86_64-linux-gnu/libruby-2.5.so.2.5 +0x9926e ruby_run_node [??:?]
libklayout_rba.so.0 +0x22b70 rba::RubyInterpreter::initialize(int&, char**, int (*)(int&, char**)) [??:?]
./klayout +0x38a7 ?? [??:0]
/lib/x86_64-linux-gnu/libc.so.6 +0x2409b __libc_start_main [libc-start.c:342]
./klayout +0x39ea ?? [??:0]
```

Then I click ok on the traceback then a dialog pops up saying 'Operation Cancelled'.

I hope that was useful.

klayoutmatthias, 2021-04-11:
@yrrapt No need to apologize - thanks for this testcase!

I'm afraid I still can't reproduce it, although I ran a some experiments with valgrind which is usually a very reliable way to detect memory corruption issues. 

I noticed the test case contains references to the sky130 library. Maybe the issue is related to them. Do you have a pointer where I can get these? 

One more question: are you using a build which is based on a recent code base? Would you mind pasting the text you see in "Help/About" here?

Thanks,

Matthias

klayoutmatthias, 2021-05-12:
@yrrapt I have some reason to hope that issue-795 relates to this issue. I'd ask you to test your case again when this bug fix is merged into the master or release as 0.27.1.

Thank,

Matthias

yrrapt, 2021-05-13:
@klayoutmatthias Great, I will try it once I've installed 0.27.1

klayoutmatthias, 2021-06-20:
@yrrapt Any updates? Do you still see this issue?

Thanks,

Matthias

klayoutmatthias, 2021-08-25:
I guess this issue is solved.

closed 2021-08-25 by klayoutmatthias