KLayout/klayout #772: attributeerror: module 'pya' has no attribute 'application'

songahnu, 2021-04-16:


klayoutmatthias, 2021-04-18:
Please provide more information

songahnu, 2021-04-18:
**Thank you for your reply
I encountered such a problem when running the following code
python3.7  klayout0.26.11**

import pya
import array
import sys
import math
import time
mw = pya.Application.instance().main_window()
layout = pya.Application.instance().main_window().current_view().active_cellview().layout() 
topcell = pya.Application.instance().main_window().current_view().active_cellview().cell#
#layout=pya.CellView.active().layout()      
#topcell=layout.cell("main")
lv = pya.Application.instance().main_window().current_view()

lv.transaction("Undo Boolean")
out = LayerInfo.new(3,0)
#out=3
#lif1=1
#lif2=2
lif1=LayerInfo.new(1,0)
li1=layout.layer(lif1,0)
lif2=LayerInfo.new(2,0)
li2=layout.layer(lif2,0)

r1=pya.Region(topcell.begin_shapes_rec(li1))
r2=pya.Region(topcell.begin_shapes_rec(li2))

selection=r1-r2
layout.clear_layer(li1)
layout.clear_layer(li2)

output=layout.layer(out)
topcell.shapes(output).insert(selection)

lv.commit()

Thank you very much
A student from China

klayoutmatthias, 2021-04-18:
The script works fine if I apply some modifications:

```
import pya
import array
import sys
import math
import time

mw = pya.Application.instance().main_window()
layout = pya.Application.instance().main_window().current_view().active_cellview().layout()
topcell = pya.Application.instance().main_window().current_view().active_cellview().cell#
#layout=pya.CellView.active().layout()
#topcell=layout.cell("main")
lv = pya.Application.instance().main_window().current_view()

lv.transaction("Undo Boolean")
out = pya.LayerInfo.new(3,0)
#out=3
#lif1=1
#lif2=2
lif1=pya.LayerInfo.new(1,0)
li1=layout.layer(lif1)
lif2=pya.LayerInfo.new(2,0)
li2=layout.layer(lif2)

r1=pya.Region(topcell.begin_shapes_rec(li1))
r2=pya.Region(topcell.begin_shapes_rec(li2))

selection=r1-r2
layout.clear_layer(li1)
layout.clear_layer(li2)

output=layout.layer(out)
topcell.shapes(output).insert(selection)

lv.commit()
```

Regards,

Matthias

songahnu, 2021-04-18:
Hi Mathias,
Thank you for your reply,The script runs in my computer(windows10), the following error will appear
"mw = pya.Application.instance().main_window()
AttributeError: module 'pya' has no attribute 'Application'" 
How can I solve it，please forgive me for being a novice.
Regards,

A student from China

closed 2021-04-22 by songahnu