KLayout/klayout #777: LEF cells not being imported?

mguthaus, 2021-04-21:
I am trying to import a DEF/LEF pair of files from the OpenLane project. I can view these fine in other tools (e.g. Magic), but the LEF files don't seem to load in klayout and show up enclosed in parens in the cell view. I am assuming that is a missing reference? THere is no 

[deflef_test.zip](https://github.com/KLayout/klayout/files/6346571/deflef_test.zip)
[image]

I am using Klayout v0.27 from the git repo commit 721a4c62

stefanottili, 2021-04-21:
There are two use cases for importing def.
One to convert it to gds and collect the gds leafcells. In this case you want the def components to be references and you don’t want to read lef macros.
The other is to view def with the lef macros. 
In 0.27.x there are options that allow you to do both.

mguthaus, 2021-04-21:
Hi @stefanottili. Yes, I'm trying to do the second: def components referencing lef macros. Where are these options to specify which use case? I don't see anything in the DEF/LEF import options -- literally, it is blank.
[image]

stefanottili, 2021-04-21:
I don't have access to 0.27.x at the moment, but in 0.26.x File->Reader Options->lefdef works.

mguthaus, 2021-04-21:
Ah, I was clicking the "Options" button in the LEFDEF import menu which doesn't work. I can get to the options in the File->Reader Options menu.

However, I still don't see any option that I'm not already using. I already specify the LEF file in the LEFDEF reader form (below). It seems that specifying it under the Reader Options->LEFDEF tab does the same thing. If I specify it in both places, I actually get a duplicate MACRO warning and it fails to load.

[image]

klayoutmatthias, 2021-04-21:
Hi Mathew,

"Import LEF/DEF" should still work, but it's legacy and I will remove it sooner or later. The reader options are taken from the current technology but I can't see how they can be empty. Maybe it's a matter of the tech setup.

For importing LEF, you have the option to load all LEFs beside the DEF or specifiy a list of LEFs to load:

[image]

So if you check the "read all LEF" box, it should do the job.

But I can confirm it doesn't. I'll debug that.

Matthias

klayoutmatthias, 2021-04-21:
The problem is the FOREIGN statement:

```
MACRO sky130_fd_sc_hd__ha_4
  CLASS CORE ;
  FOREIGN sky130_fd_sc_hd__ha_4 ;
  ORIGIN  0.000000  0.000000 ;
  SIZE  9.200000 BY  2.720000 ;
  SYMMETRY X Y R90 ;
  SITE unithd ;
...
END sky130_fd_sc_hd__ha_4
```

With this option, the reader will always assume the LEF cells are to come from some external source.

@stefanottili You recall our discussion? I think this was the outcome. But there is still time to change it :)

There is an option to create LEF geometry, but it only applies to cells not having FOREIGN. Macros with FOREIGN are always translated into ghost cells.

The idea of the FOREIGN implementation in KLayout is that you can use a GDS to supply the actual and detailed layouts for the gate cells. 

For adding the library cells, you can either use a script or on of the "strm2x" tools. You'll need a library containing the layouts of the standard cells. Here I'm using "strm2gds" and I assume the standard cell lib is called "sky130_lib.gds":

```
strm2gds spm.placement.def+sky130_lib.gds out.gds --blend-mode=1
```

"blend-mode=1" will instruct the reader to substitute cells with identical names from "sky130_lib.gds". 

With this approach, unused library cells will show up as new top cells in the output layout however. 

Scripting allows coding all these options in a flexible way and tailored to your specific requirements.

Best regards,

Matthias

klayoutmatthias, 2021-04-21:
I'd still like to know why the setup was empty on "Import LEF/DEF". Are you using a specific technology? Is this option empty as well with the default technology?

Thanks,

Matthias

mguthaus, 2021-04-21:
@klayoutmatthias I am not sure why it was empty. I updated versions and it seems that it was a temporary issue. I also cleared my klayoutrc, so perhaps there was some issue there.

Looking at the [LEF/DEF reference](http://www.ispd.cc/contests/18/lefdefref.pdf), it seems that FOREIGN should specify the GDs file name *if you want to convert to GDS*. Specifically, "Specifies the foreign (GDSII) structure name to use when placing an
instance of the macro." It sounds like there should still be an option to display the LEF abstracted view even with a FOREIGN statement in the LEF file.

stefanottili, 2021-04-21:
The current klayout master behavior is not what other tools do.

Let me reiterate my understanding:

1) reading lef macros: ignore foreign completely, create cell with lef geometry

2a) no foreign or foreign == macro name: reference name is macro name

2b) macro name != foreign. 
This is rare and warrants an info or warning. 
Usually you see this as an unintended mistake copying and renaming an abstract , forgetting to rename the foreign property.
It can be done in two ways:
2b1) use foreign instead of macro name for the reference (treat foreign just as a rename)
2b2) create struct with macro name that then contains a reference to foreign
is what I’ve seen since cell3 in 1995
I’ve never seen macro with foreign offset != macro offset and the few times I’ve seen orientation being used, it was always N.

klayoutmatthias, 2021-04-21:
Actually this option is there internally, but I was not sure what's a good set of options for users.

Put the foreign name != macro name problem aside, these are the three options available internally:

1. read LEF geometry unless FOREIGN is present. In case of FOREIGN produce a ghost cell for later insertion of a GDS cell.
2. always read LEF geometry and ignore FOREIGN
3. always produce a ghost cell and ignore LEF goemetry

Currently the "read LEF geometry" option is wired to 1 (on) and 3 (off). But I can map it to 2 (on) and 1 (off) if that hits the spot. I just recall the three options were too confusing :)

Matthias

P.S. I have too little experience myself in what other tools do with LEF/DEF ...

P.P.S. you can access the second option (which I assume is the desired one) by this script already:

```
# Reads a DEF file with all LEF's next to it from path and opens it in a new panel

path = "path-to-your.def"

import pya

mw = pya.MainWindow.instance()
cv = mw.create_layout(True)
ly = cv.layout()

options = pya.LoadLayoutOptions()
lefdef_options = options.lefdef_config
lefdef_options.macro_resolution_mode = 1   # Option 2: Ignore FOREIGN and use LEF layout
options.lefdef_config = lefdef_options
ly.read(path, options)

cv.cell = ly.top_cell()
cv.view().add_missing_layers()
```

stefanottili, 2021-04-21:
There should only one toggle switch „generate cell for macro“ or not.
Either you want to see the macro geometry, or you want to collect the referenced cell from a different gds/oasis file.

And then you have to deal with the very rare macro != foreign for both cases.

mguthaus, 2021-04-21:
Is it possible to make the resolution mode a menu option? I would suggest:

Default: If FOREIGN is present, look for the GDS, if not fall back to LEF.
LEF: Always select LEF:

Often these DEF files are viewed just as intermediate debugging during placement or routing. Having to script something to view something one time is a bit of a pain.

stefanottili, 2021-04-21:
Either you want to reference lef or you want to reference gds.

This is completely independent of having a or the contents of the foreign statement.

mguthaus, 2021-04-22:
Not quite independent. If you specify to look for the GDS and it is specified by a FOREIGN statement, there should be some logical fallback. That would be the LEF which is definitely available. 

Other than that, I do agree.

mguthaus, 2021-04-22:
Actually, it should do this without a FOREIGN statement as well.

klayoutmatthias, 2021-05-02:
The new 0.27 version does have an option to read LEF geometry always:

[image]

However, there is no user interface yet to use a GDS file for resolving the LEF macros. The "strm2xyz" tools however have such an option already: "--lefdef-lef-layouts lib.gds,lib2.gds,...".

Matthias

klayoutmatthias, 2021-09-24:
I'll close this issue as I think there are enough options for now :) 

If someone wants to have more discussion here, please reopen.

Matthias

closed 2021-09-24 by klayoutmatthias