KLayout/klayout #783: SiEPIC installation

ansingh07, 2021-04-30:
Hi,

I need SiEPIC package for AIM photonics pdk and for that purpose I used the instruction [here](https://github.com/SiEPIC/SiEPIC-Tools/wiki/Installation) on my Mac OS X Catalina.

With Anaconda3 option: KLayout is crashing [error: DYLD library mission]. See attached picture. I tried to troubleshoot with answers in KLayout discussion forum in GitHub but nothing worked for me. 

With HomeBrew option: KLayout is installed and running but it cannot install numpy (tried several times already).

I tried out different version of KLayout already, but no success. It would be great if I can get any help on this. 

thanks
[image]

Kazzz-S, 2021-04-30:
Hi,

Regarding `LW*qt5Ana3*.dmg`, you need Ruby 2.5.x installed, which looks missing in your Anaconda3 environment.

Try
```
$ conda install ruby
```
Then,
```
$ conda list | grep package_name
```
should report something like
```
qt      5.9.7   h468cd18_1
python  3.8.2   hc70fcce_0
ruby    2.5.1   h7107397_0
```

Kazzz-S

ansingh07, 2021-04-30:
Hi,

Thanks for your quick response. I did this step already as a first step of the "Anaconda3User-ReadmeFirst.txt". I ran that "conda install ruby" command again. I would like to point out that it takes forever to install ruby as system finds conflicts and try to solve them. What I don’t know if it installs ruby correctly or not. With "conda list | grep package_name" it doesn't give any output.

Kazzz-S, 2021-04-30:
Hi,

I understood your problem. It's weird.
However, the error message attached below indicates that the Ruby library is not loaded.
[image]

What I can suggest is to clean reinstall Anaconda3.

Meanwhile, I have just built the `0.26.12` maintenance version.
This DMG will not solve your problem promptly, but try it if you like.
https://www.dropbox.com/s/slnk9k9z8k7tg92/LW-klayout-0.26.12-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg?dl=0 
(After a while, this link may be lost.)

[image]
Kazzz-S

ansingh07, 2021-04-30:
Hi, On a suggestion of my colleague, I tried again with HomeBrew and used

brew install ruby@2.7
brew install python@3.8
brew install qt5 

then installed KLayout. Now its working for me and able to install SiEPIC tool and AIM Photonics library successfully.