KLayout/klayout #784: Mac DMGs for 0.26.12

Kazzz-S, 2021-05-01:
Dear @klayoutmatthias,

Thank you for the maintenance release.

As usual, different DMG packages of **KLayou 0.26.12 for Mac** are available as below ([0.26.12 Dropbox Link](https://www.dropbox.com/sh/copwmwlw1b6heur/AACx2FoDiVra-Lg1LWNqJt_0a?dl=0)).
**Please note that _pymod_ is not included in those packages.**

Another news is that Stefan and I are working to build KLayout for the Apple Silicon M1 chip, which will be available from
0.27 onward.

## DMG file naming convention ##
**0.26.12 for Catalina** is taken as an example here.
[image]

where
(a) Type of package:  ST-: standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt5 type:  qt5MP: MacPorts; qt5Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:   sys: OS-bundled; mp27: Ruby2.7 from MacPorts; hb27: Ruby2.7 from Homebrew; ana3: Ruby2.5 from Anaconda3
(f) **P**ython type: sys: OS-bundled; mp38: Python 3.8 from MacPorts; hb38: Python 3.8 from Homebrew; ana3: Python3.8 from Anaconda3

## For the primary platform, macOS Catalina (10.15) ##
**LW- and HW- packages use Python 3.8 in MacPorts, Homebrew and Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
1|Catalina (10.15)|ST-klayout-0.26.12-macOS-Catalina-1-qt5MP-RsysPsys.dmg|Homebrew|OS-bundled|OS-bundled|(1)|
2|Catalina (10.15) with MacPorts dev. env.|LW-klayout-0.26.12-macOS-Catalina-1-qt5MP-Rmp27Pmp38.dmg|MacPorts|MacPorts|MacPorts|(2)|
3|Catalina (10.15) with Homebrew dev. env.|LW-klayout-0.26.12-macOS-Catalina-1-qt5Brew-Rhb27Phb38.dmg|Homebrew|Homebrew|Homebrew|(3)|
4|Catalina (10.15) with Anaconda3 dev. env.|LW-klayout-0.26.12-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(4)|
5|Catalina (10.15)|HW-klayout-0.26.12-macOS-Catalina-1-qt5Brew-RsysPhb38.dmg|Homebrew|OS-bundled|Homebrew|(5)|

(1) shares OS-bundled Ruby2.6 and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).
(2) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(3) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(4) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(5) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.8 from Homebrew are embedded, hence, heavy.

## For the latest macOS Big Sur (11.3) ##
Five macOS DMGs for **Big Sur** are ready for experimental purposes.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 12.1.0.

**LW- and HW- packages use Python 3.8 in MacPorts, Homebrew and Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
21|BigSur (11.2)|ST-klayout-0.26.12-macOS-BigSur-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(21)|
22|BigSur (11.2) with MacPorts dev. env.|LW-klayout-0.26.12-macOS-BigSur-1-qt5MP-Rmp27Pmp38.dmg|MacPorts|MacPorts|MacPorts|(22)|
23|BigSur (11.2) with Homebrew dev. env.|LW-klayout-0.26.12-macOS-BigSur-1-qt5Brew-Rhb27Phb38.dmg|Homebrew|Homebrew|Homebrew|(23)|
24|BigSur (11.2) with Anaconda3 dev. env.|LW-klayout-0.26.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(24)|
25|BigSur (11.2)|HW-klayout-0.26.12-macOS-BigSur-1-qt5Brew-RsysPhb38.dmg|Homebrew|OS-bundled|Homebrew|(25)|

(21) **experimental**; shares OS-bundled Ruby2.6 and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).
(22) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(23) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(24) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(25) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.8 from Homebrew are embedded, hence, heavy.

## For previous OS ##
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Mojave (10.14)|ST-klayout-0.26.12-macOS-Mojave-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1p)|
7|HighSierra ( 10.13)|ST-klayout-0.26.12-macOS-HighSierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1p)|
8|Sierra (10.12)|ST-klayout-0.26.12-macOS-Sierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1p)|
9|ElCapitan (10.11)|ST-klayout-0.26.12-MacOSX-ElCapitan-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled| (1p) |
x|Yosemite (10.10)|has been dropped due to the compiler issue reported in #360.|

(1p) shares OS-bundled Ruby[2.0 ~ 2.3] and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).

Please close this ticket after publishing the packages.
With warm regards,
Kazzz-S

klayoutmatthias, 2021-05-02:
The images have been uploaded to the server. Thanks for providing them!

Best regards,

Matthias

Kazzz-S, 2021-05-08:
## Notes on Homebrew Versions (2021-05-08) ##

Users who are new to KLayout with Homebrew environment may not be able to start `KLayout 0.26.12` from the two DMGs below,
   * LW-klayout-0.26.12-macOS-BigSur-1-qt5Brew-Rhb27Phb38.dmg
   * LW-klayout-0.26.12-macOS-Catalina-1-qt5Brew-Rhb27Phb38.dmg

if newly installed `qt@5`, `ruby@2.7`, and `python@3.8` by:
```
$ brew install qt@5
$ brew install ruby@2.7
$ brew install python@3.8
```

And have the Homebrew environment similar to the one shown below (a sample case of Intel Mac).
```
(base) BigSur2{kazzz-s} opt (1)% pwd
/usr/local/opt

(base) BigSur2{kazzz-s} opt (2)% ll
total 0
drwxrwxr-x 22 kazzz-s admin 704 May  8 09:39 .
drwxr-xr-x 14 root    wheel 448 May  4 21:56 ..
-rw-r--r--  1 kazzz-s admin   0 Nov 19 20:47 .keepme
lrwxr-xr-x  1 kazzz-s admin  25 Apr  1 06:40 binutils -> ../Cellar/binutils/2.36.1
lrwxr-xr-x  1 kazzz-s admin  19 Mar 12 18:56 gdbm -> ../Cellar/gdbm/1.19
lrwxr-xr-x  1 kazzz-s admin  25 Mar  2 18:50 libmpdec -> ../Cellar/mpdecimal/2.5.1
lrwxr-xr-x  1 kazzz-s admin  23 Dec 22 21:00 libyaml -> ../Cellar/libyaml/0.2.5
lrwxr-xr-x  1 kazzz-s admin  25 Mar  2 18:50 mpdecimal -> ../Cellar/mpdecimal/2.5.1
lrwxr-xr-x  1 kazzz-s admin  28 Apr  1 06:41 openssl -> ../Cellar/openssl@1.1/1.1.1k
lrwxr-xr-x  1 kazzz-s admin  28 Apr  1 06:41 openssl@1.1 -> ../Cellar/openssl@1.1/1.1.1k
lrwxr-xr-x  1 kazzz-s admin  29 Feb  5 21:19 pkg-config -> ../Cellar/pkg-config/0.29.2_3
lrwxr-xr-x  1 kazzz-s admin  29 Feb  5 21:19 pkgconfig -> ../Cellar/pkg-config/0.29.2_3
lrwxr-xr-x  1 kazzz-s admin  27 May  4 20:41 python@3.8 -> ../Cellar/python@3.8/3.8.10
lrwxr-xr-x  1 kazzz-s admin  21 May  8 09:39 qt5 -> ../Cellar/qt@5/5.15.2
lrwxr-xr-x  1 kazzz-s admin  21 May  8 09:39 qt@5 -> ../Cellar/qt@5/5.15.2
lrwxr-xr-x  1 kazzz-s admin  22 Dec 22 21:00 readline -> ../Cellar/readline/8.1
lrwxr-xr-x  1 kazzz-s admin  24 May  8 09:36 ruby@2 -> ../Cellar/ruby@2.7/2.7.3
lrwxr-xr-x  1 kazzz-s admin  24 May  8 09:36 ruby@2.7 -> ../Cellar/ruby@2.7/2.7.3
lrwxr-xr-x  1 kazzz-s admin  23 May  1 08:34 sqlite -> ../Cellar/sqlite/3.35.5
lrwxr-xr-x  1 kazzz-s admin  23 May  1 08:34 sqlite3 -> ../Cellar/sqlite/3.35.5
lrwxr-xr-x  1 kazzz-s admin  25 Feb 25 19:03 tcl-tk -> ../Cellar/tcl-tk/8.6.11_1
lrwxr-xr-x  1 kazzz-s admin  18 Dec 22 21:01 xz -> ../Cellar/xz/5.2.5
```

A simple workaround is to make the following two symbolic links.
```
lrwxr-xr-x  1 kazzz-s admin   8 May  8 09:42 ruby -> ruby@2.7
lrwxr-xr-x  1 kazzz-s admin   4 May  8 09:43 qt -> qt@5
```

Another option is to reinstall the 2nd-built DMGs:
  * `LW-klayout-0.26.12-macOS-Catalina-2-qt5Brew-Rhb27Phb38.dmg` => [0.26.12 Dropbox Link](https://www.dropbox.com/s/6be5vixoupwdkk5/LW-klayout-0.26.12-macOS-Catalina-2-qt5Brew-Rhb27Phb38.dmg?dl=0)
    MD5=f872dd211c9da7868d55fc014090ba3f

  * `LW-klayout-0.26.12-macOS-BigSur-2-qt5Brew-Rhb27Phb38.dmg` => [0.26.12 Dropbox Link](https://www.dropbox.com/s/qazjf3irozwj9vr/LW-klayout-0.26.12-macOS-BigSur-2-qt5Brew-Rhb27Phb38.dmg?dl=0)
    MD5=8aab63c9cf7aa2db0f06d779b2d14731

Sorry about the inconvenience caused.
Kazzz-S

klayoutmatthias, 2021-05-29:
I'll close it as the images are on the server now :)

closed 2021-05-29 by klayoutmatthias