KLayout/klayout #786: Mac DMGs for 0.27

Kazzz-S, 2021-05-02:
Dear @klayoutmatthias,

Thank you for making the major releases of 0.27.
I have branched off ee984f6b and updated the build system for Mac to support the Apple Silicon chip.
I could successfully build five different DMGs for Catalina, as you see in this [Dropbox link (1).](https://www.dropbox.com/sh/qcvwqqecp6x88cm/AACTZpnP0N_D3pACDf_VAQpxa?dl=0)
However, QA tests failed in a very early stage, as you see in these logs files [Dropbox link (2).](https://www.dropbox.com/sh/1x08hs4vcbmvkv7/AAD3dcbwp72A4dSiBTkHDigsa?dl=0)
(Please note that I will delete/replace these files after a while.)

Could you please investigate the possible causes?
Regards,
Kazzz-S

Kazzz-S, 2021-05-10:
Some more findings and experiment results are here.

1. The error below reproduces when starting the `ut_runner` program with the `-x 'pymod,pya'` option.
```
    Unable to load plugin tests: /path-to-build-dir/qt5Brew.build.macos-Catalina-release-Rhb27Phb38.macQAT/view_25d_tests.ut
```

2. According to Stefan (he has tried with the master branch), the same error flags on M1 Mac, too.

3. `view_25d_tests.ut` is a new unit test in 0.27 (0.26.x does not have it).

4. From `unit_test_main.cc`, I understand that `ut_runner` **dlopen()s**  binaries with `*.ut` extension in the same directory as `view_25d_tests.ut` one by one.
   - I renamed it to exclude, then `ut_runner` could start.
   - _Do I need to add some directories to **DYLD_LIBRARY_PATH**?_

5. `ut_runner` takes a longer time than when testing 0.26.x.
    It looks like the QA test has become more rigorous :-)

6. `ut_runner` reports many error messages. The log file is attached [here](https://github.com/KLayout/klayout/files/6455133/QATest_fca3bb7f_2021_0511_0600__qt5Brew.build.macos-Catalina-release-Rhb27Phb38.macQAT.zip).

Thanks
Kazzz-S

klayoutmatthias, 2021-05-11:
HI kazzz,

thanks for this information.

You need to add "lay_plugins" to DYLD_LBRARY_PATH to solve the problem
with view_25d_tests.ut.

I found a host service that offers a private M1 Mac with remote access
for 15 Euro in 24h increments. I'll give that a try a try to debug the
problems Stefan had with M1, but it will take a while until I find 24
hours for this topic :)

Best regards,

Matthias

Kazzz-S, 2021-05-12:
Hi Matthias,

Thank you for your instruction.

1. [With this change](https://github.com/Kazzz-S/klayout/commit/2e9f3bc2deb72cc7afbde25e33de948e47fed627), I could normally start the `rt_runner` program :-)
2. Rebuilt five DMGs for Catalina are in this [Dropbox Link (1)](https://www.dropbox.com/sh/9vmoh89gnioi0af/AABM6dGmkTrxWmI9AmNrShfoa?dl=0). Some of them should work on BigSur if the environments match.
3. QA test log files are in this [Dropbox Link (2)](https://www.dropbox.com/sh/a33et8cq9oet1yo/AAD9Xt3L9ysR5jz_xUx2c6CQa?dl=0).

Regards,
Kazzz-S

Kazzz-S, 2021-05-13:
I've uploaded another five DMGs built using a Big Sur (Intel Mac) virtual machine and their QA test log files to the same Dropbox Links shown above.

Kazzz-S

Kazzz-S, 2021-06-03:
Closed since 0.27.1 has been released. 
This refs #819.

closed 2021-06-03 by Kazzz-S