KLayout/klayout #809: with_density goes outside the boundary of the input layer

ahmednofal, 2021-05-25:
Thank you for your work on this project. I am using `without_density` function to retrieve the polygons in a layer that do not achieve a certain density. My input layer has a bounding box that `without_density` function-produced tiles go out of.
 Blue is the tiles produced by `without_density`, red is the input layer to the `without_density` function.
[image]
```
     input_layer = input(wild_card)
     print("\n#{input_layer.bbox}")
     input_layer = input_layer.and(chip_boundary)
     print("\n#{input_layer.bbox}")
     bad_density = input_layer.without_density(0.33..0.57, tile_size(700.um), tile_step(70.um))
     good_density = input_layer.with_density(0.33..0.57, tile_size(700.um), tile_step(70.um))
     target("output.gds")
     input_layer = input(wild_card)
     input_layer.output(wild_card)
     bad_density.output(arbitrary_layer_num)
     good_density.output(arbitrary_layer_num2)
```

with a bounding box of `(7.92,7.9;3592.5,5192.45)`
It is not clear which layer(and consequently which boundary) is used for the base layer(which you divide the area of the polygon on), which would normally be the boundary layer.  I tried using `tile_boundary` but it turns out that this options INCREASES the boundary and can not reduce it.

klayoutmatthias, 2021-05-26:
@ahmednofal If this is a support request, please consider using the forum for this purpose: https://www.klayout.de/forum/

"tile_boundary" will just add more layers for the bounding box computation, not confine the area computation. So it's clear that this will expand your bounding box.

The tiles are chosen so that each tile gets at least a tiny bit of the layout, hence the furthest tiles will just nibble at the border of the layout. Currently, everything outside is considered zero density, hence, the density falls off for the outer tiles and a low density error is triggered.

It's possible to adjust the implementation so only the area inside "tile_boundary" or the bounding box is checked - i.e. a mask rather than assume zero density for the outside part. I'd like to leave this request open to a discussion here or in the forum.

You can also compute the number of tiles and the origin so no tiles are outside your chip's area:

```
input_layer = input(1, 0)

bbox = input_layer.bbox

puts "bbox=" + bbox.to_s

step = 70.um
size = 700.um

density = 0.2 .. 0.57 # for my pattern

# compute minimum number of tiles and a good offset
nx = ((bbox.width - (size - step)) / step).to_i
ny = ((bbox.height - (size - step)) / step).to_i
ox = bbox.left + (size - step) / 2
oy = bbox.bottom + (size - step) / 2
puts "nx=" + nx.to_s + ", ny=" + ny.to_s
puts "ox=" + ox.to_s + ", oy=" + oy.to_s

bad_density = input_layer.without_density(0.33..0.57, tile_size(size), tile_step(step), tile_origin(ox, oy), tile_count(nx, ny))
good_density = input_layer.with_density(0.33..0.57, tile_size(size), tile_step(step), tile_origin(ox, oy), tile_count(nx, ny))
bad_density.output(100, 0)
good_density.output(101, 0)
```

However, because the tile count is rounded down, there will be a small area of max 70µm width and height at the top and right side which is not checked. The integrated solution will avoid this.

Matthias

klayoutmatthias, 2021-05-29:
The solution consists of having a mode switch which tells "with_density" to change from "zero density outside" to "ignore outside". The keyword is "padding_ignore":

```
result = layer.with_density(..., padding_ignore)
```

Matthias

klayoutmatthias, 2021-06-01:
This feature is released with 0.27.1.

closed 2021-06-01 by klayoutmatthias