KLayout/klayout #810: Python, os.path.realpath returns extra '/

lukasc-ubc, 2021-05-26:
Hi Matthias,

On klayout-HW-klayout-0.27-macOS-BigSur-1-qt5Brew-RsysPhb38, in the Macro window in Python the following code returns something different than on the 0.26 versions, namely extra prepended characters:

```
> import os
> os.path.realpath(__file__)
/'/Users/lukasc/Documents/GitHub/SiEPIC-Tools/klayout_dot_config/tech/GSiP/pymacros
```

in HW-klayout-0.26.12-macOS-BigSur-1-qt5Brew-RsysPhb38.dmg, the same returns:
```
/Users/lukasc/Documents/GitHub/SiEPIC-Tools/klayout_dot_config/tech/GSiP/pymacros
```

lukasc-ubc, 2021-05-26:
Furthermore, breakpoints don't seem to work:

[image]

lukasc-ubc, 2021-05-26:
also,

for Library.register, the documentation states that it replaces the library. https://www.klayout.de/doc-qt5/code/class_Library.html#method24

self.register(library)

However, if I run the library multiple times, it shows up repeated:

[image]

In 0.26.12, it replaced correctly.  Furthermore, documentation says that this and other features were introduced in 0.27, but they were already in 0.26.

lukasc-ubc, 2021-06-12:
@klayoutmatthias 

- the path problem seems to be fixed in 0.27.1 (HW version on OSX)
- the breakpoint problem seems to be fixed in 0.27.1 (HW version on OSX)

but:
- the Library.register function doesn't do the replacement

regards,
Lukas

lukasc-ubc, 2021-07-02:
@klayoutmatthias 
The os.path problem is back in 0.27.2

I get the issue:

`/'/Users/lukasc/Documents/GitHub/SiEPIC-Tools/klayout_dot_config/tech/GSiP/pymacros
`

klayoutmatthias, 2021-07-06:
Hi Lucas,

I tried first to reproduce the "library.register" problem, but I can't confirm that (Linux).

This is what works for me:

```
lib_a = RBA::Library::new
lib_a.layout.add_cell("LIB_A")
lib_a.register("ISSUE_810_LIB")

lib_b = RBA::Library::new
lib_b.layout.add_cell("LIB_B")
lib_b.add_technology("si4all")
lib_b.register("ISSUE_810_LIB")

# register another library for tech "si4all" which replaces lib_b
lib_c = RBA::Library::new
lib_c.layout.add_cell("LIB_C")
lib_c.add_technology("si4all")
lib_c.register("ISSUE_810_LIB")

# register another library for "no tech" which replaces lib_a
lib_x = RBA::Library::new
lib_x.layout.add_cell("LIB_X")
lib_x.register("ISSUE_810_LIB")
```

This will correctly register two libraries "ISSUE_810_LIB", one for "si4all" technology, the other unspecific. The first two "register" calls are replaced.

klayoutmatthias, 2021-07-06:
I do have only Catalina, so I tried to reproduce the problem with os.path.

I used HW-klayout-0.27.2-macOS-Catalina-1-qt5Brew-RsysPhb38.dmg without success:

[image]

So maybe it's a problem with BigSur?

lukasc-ubc, 2021-07-06:
@Kazzz-S could you please check these two issues, if they they show up on your OSX installations?

thanks

Kazzz-S, 2021-07-06:
Hi,

In my environment...

## Catalina ##
* HW-klayout-0.27.2-macOS-Catalina-1-qt5Brew-RsysPhb38.dmg
[image]
[image]

## Big Sur ##
* HW-klayout-0.27.2-macOS-BigSur-1-qt5Brew-RsysPhb38.dmg
[image]
[image]

Kazzz-S

lukasc-ubc, 2021-07-07:
Could you please try with a folder that has a space in the name? This could be causing an issue. Thank you. 

Lukas Chrostowski
Professor, Electrical and Computer Engineering
University of British Columbia
Quantum Matter Institute
Member, College of the Royal Society of Canada
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca
https://qmi.ubc.ca/about/directory/lukas-chrostowski

Kazzz-S, 2021-07-07:
Is this a useful test?

```
#--------------------------
# Issue-810
#
# File: Issue810-2.py
#--------------------------
import os
import platform

# The directory structure is as shown below.
#   (base) BigSur2{kazzz-s} GitHub810 (1)% pwd
#   /Users/kazzz-s/Gitwork/ForumKLayout/GitHub810
#   (base) BigSur2{kazzz-s} GitHub810 (2)% ll
#   total 0
#   drwxr-xr-x  4 kazzz-s staff 128 Jul  7 18:14  .
#   drwxr-xr-x 18 kazzz-s staff 576 Jul  7 18:03  ..
#   drwxr-xr-x  4 kazzz-s staff 128 Jul  7 18:03 'Folder With Space'
#   lrwxr-xr-x  1 kazzz-s staff  17 Jul  7 18:14  symlink -> 'Folder With Space'

myFileReal = os.environ['HOME'] + "/GitWork/ForumKLayout/GitHub810/Folder With Space/Issue810-2.py"
myFileSymL = os.environ['HOME'] + "/GitWork/ForumKLayout/GitHub810/symlink/Issue810-2.py"

print( "### My Platform ###" )
(System, Node, Release, Version, Machine, Processor) = platform.uname()
print( "System    = %s" % System )
print( "Node      = %s" % Node )
print( "Release   = %s" % Release )
print( "Version   = %s" % Version )
print( "Machine   = %s" % Machine )
print( "Processor = %s" % Processor )

print( "" )
print( "### Real Path with Spaces ###" )
path1=os.path.realpath(myFileReal)
print(path1)

path2=os.path.dirname(os.path.realpath(myFileReal))
print(path2)

print( "" )
print( "### Symbolic Link Path  ###" )
path3=os.path.realpath(myFileSymL)
print(path3)

path4=os.path.dirname(os.path.realpath(myFileSymL))
print(path4)

# End of file
```

[image]

[image]

Kazzz-S

seanlam97, 2022-01-07:
This seems to be happening on Windows as well but uses path to KLayout installation. Running Windows 11 with KLayout 0.27.5

[image]