KLayout/klayout #811: Paths, boxes, etc. instantly converted to polygons when placed

karl-johnson, 2021-05-27:
I can't determine if this is a bug or some sort of setting I have selected, but I've looked though the documentation and issues and I haven't seen any other reports of this. This also comes and goes sporadically, and I've had it happen on 0.26.9, 0.26.12, and 0.27. 

In essence, whenever I place _any_ object, it is immediately converted to a polygon, making me unable to edit it other than adjusting the points comprising its boundary. For example, here is me placing a box:

[image]

I click once to start the box, click again to finish it, then press escape. Double clicking on the box then yields

[image]

Important to note is that not all objects in the layout are converted - all existing paths/boxes are still editable as such, but any new ones I place are only able to be edited as polygons. There is nothing in the Log Viewer indicating an error when I do this. Yesterday, when this would happen, I noticed that it would only occur on certain layers, and when I would successfully make a non-polygon object on a layer that didn't have this issue, the issue would suddenly disappear on _all layers_ (for a while). Today, however, all objects I place have this issue.  Is this a setting somewhere in the software that I'm missing, or is this a bug?

lukasc-ubc, 2021-05-28:
check that it the button says “Add”

instead of “Merge”

klayoutmatthias, 2021-05-29:
Exactly :)

karl-johnson, 2021-05-30:
What a simple mistake, I can't believe I didn't see that.

closed 2021-05-30 by karl-johnson