KLayout/klayout #814: runing slow

wolfgold, 2021-05-31:
I compiled 0.24 klayout by myself with VS2010, but the running speed of the version I compiled is much slower than the 0.24 version provided. Is there any special compilation option.

klayoutmatthias, 2021-06-01:
Did you use a debug build maybe?

Please don't expect me to give a reasonable answer without providing any details.

BTW: 0.24 is entirely outdated.

Matthias

wolfgold, 2021-06-03:
@klayoutmatthias Thank you very much for your reply. I have found the reason. You can just select O2 in the compilation option.I extracted part of the 0.24 source code and used Visual Studia's STL directly instead of using STLPort. WIll this affects the speed of operations such as Boolean and merge.

wolfgold, 2021-06-07:
it's STLPort's effects.

closed 2021-06-07 by wolfgold