KLayout/klayout #815: KLayout 0.27 zoom bug

refob, 2021-05-31:
I'm working with the current KLayout 0.27 version on CentOS7 and Win10. The zoom function using the right mouse button does not work as expected.  Zoomout using the RMB works as expected in the Main Layouut and the Navigator window. Zoomin using the RMB works only in the Navigator window as expected. In the Main Layout window using the RMB to zoomin always executes a "fit window".

I used the binaries provided on the official website to test this.

stefanottili, 2021-05-31:
RMB is using gestures for three functionalities:
move down right => zoom in
move upper right => win fit
move left => zoom out
The last two also draw a rectangle, so it’s not obvious visually that they don’t do zoom in.
A feature request would be to change the cursor for these two cases to make the functionality more obvious.

stefanottili, 2021-05-31:
Only the main layout window uses gestures, in navigator rmb always zooms in.

refob, 2021-05-31:
Many thanks for the hint. I misunderstood the manual here.

Just one personal comment: it would be more logical for me, if lower left+drag upper right would zoomin and upper right+drag lower left would zoom out. That would be a more natural move for my hand. Most other layout editors work that way.

klayoutmatthias, 2021-06-01:
Ok, the navigator could use the win fit/zoom out gesture too. But as it's static, to me it does not seem to need to zoom fit or zoom out and there is still F2.

BTW: "most layout editors work this way ..." - how many did you check?

Here is for example:
* KiCAD PCB module: left mouse button + drag = zoom in, right mouse button + drag = zoom out
* Virtuoso layout: rmb always zoom in
* Calibre DRV: rmb move right + down or up: zoom in, rmb move left + down or up: zoom out

I don't see a common scheme here.

Matthias

klayoutmatthias, 2021-06-01:
I don't think this is a bug. I would like to close this issue.

stefanottili, 2021-06-01:
It's a feature, not a bug. But it can be unexpected for the uninitiated user.

A feature request would be to make the gestures more obvious by changing the "draw rectangle for zoom in" into something else for win fit/zoom out.
NLView changes the cursor to "win fit" text or displays and arrow and x2, x3, x4 when zooming out.

stefanottili, 2024-08-23:
Revisiting this three year old feature request:

It would be very helpful for the uninitiated user if the "fit" and "zoom out" gestures would change the cursor to a text string reflecting what they're doing when being initiated. Having the same "draw rectangle" feedback doesn't distinguish between the three functionalities. nlview went as far as having x2, x3, x3 depending on how far the "zoom out" gesture moved the mouse from the initial mouse click.