KLayout/klayout #816: KLayout 0.27: package "tf_import 1.0" dies with error message

refob, 2021-05-31:
I installed klayout-0.27-0.x86_64.rpm on CentOS7 and used the Salt Package Manager to install the the package "
tf_import 1.0". Then I ran this command:
---
klayout -rd tf_file=df2/NCSU_Techlib_tsmc02_laydefs.tf -rd lyp_file=out.lyp fifo.gds 

Which outputs this error:
--
ERROR: /home/reinhard/.klayout/salt/tf_import/ruby/import_tf.rb:70: uninitialized constant TechfileToKLayout::StringIO
  /home/reinhard/.klayout/salt/tf_import/ruby/import_tf.rb:70:in `read_skill_file_as_ruby_expr'
  /home/reinhard/.klayout/salt/tf_import/ruby/import_tf.rb:195:in `import_techfile'
  /home/reinhard/.klayout/salt/tf_import/macros/import_tf_standalone.rb:35:in `<module:TechfileToKLayout>'
  /home/reinhard/.klayout/salt/tf_import/macros/import_tf_standalone.rb:24:in `<main>' (class NameError)

Executing File>Import Cadence Techfile in the gui showed this error:
--
ERROR: /home/reinhard/.klayout/salt/tf_import/ruby/import_tf.rb:70: uninitialized constant TechfileToKLayout::StringIO
  /home/reinhard/.klayout/salt/tf_import/ruby/import_tf.rb:70:in `read_skill_file_as_ruby_expr'
  /home/reinhard/.klayout/salt/tf_import/ruby/import_tf.rb:195:in `import_techfile'
  /home/reinhard/.klayout/salt/tf_import/macros/import_tf.lym:32:in `<module:TechfileToKLayout>'
  /home/reinhard/.klayout/salt/tf_import/macros/import_tf.lym:17:in `<main>' (class NameError)

klayoutmatthias, 2021-06-01:
This does not belong here. It's not a KLayout bug. Please take it there: https://github.com/klayoutmatthias/tf_import

Matthias

refob, 2021-06-02:
Move to https://github.com/klayoutmatthias/tf_import