KLayout/klayout #817: KLayout 0.27: LEF/DEF reader layer map - purpose DRAWING ignored

refob, 2021-06-01:
I cannot post anything to the Klayout forum, so I will report the issue here. I'm using this ruby code to read a LEF/DEF-based design using "klayout -rm myscript.rb":
--
options = RBA::LoadLayoutOptions::new
lefdef_options = RBA::LEFDEFReaderConfiguration::new
lefdef_options.dbu = 0.001
lefdef_options.macro_resolution_mode  = 1
lefdef_options.lef_files = ["../lef/cmos180.lef","../lef/cmos180osu.lef"]
lefdef_options.map_file = "../klayout/cmos180.layer_map"
options.lefdef_config = lefdef_options
app = RBA::Application::instance
app.main_window.load_layout("input/def/fifo.def", options, "cmos180", 2)
--

KLayout reports these problems:
--
Warning: Reading layer map file /home/reinhard/work/input/def/../klayout/cmos180.layer_map, line 1: purpose DRAWING ignored for layer metal1
Warning: Reading layer map file /home/reinhard/work/input/def/../klayout/cmos180.layer_map, line 3: purpose BOUNDARY ignored for layer metal1
Warning: Reading layer map file /home/reinhard/work/input/def/../klayout/cmos180.layer_map, line 4: purpose DRAWING ignored for layer via
Warning: Reading layer map file /home/reinhard/work/input/def/../klayout/cmos180.layer_map, line 5: purpose BOUNDARY ignored for layer via
Warning: Reading layer map file /home/reinhard/work/input/def/../klayout/cmos180.layer_map, line 6: purpose DRAWING ignored for layer metal2
Warning: Reading layer map file /home/reinhard/work/input/def/../klayout/cmos180.layer_map, line 8: purpose BOUNDARY ignored for layer metal2
....
---

The file "cmos180.layer_map" contains these lines:
---
metal1              drawing                     49  0
metal1              pin                         81  0
metal1              boundary                    49  5
via                 drawing                     50  0
via                 boundary                    50  5
metal2              drawing                     51  0
metal2              pin                         82  0
metal2              boundary                    51  5
...
----

In KLayout I can see that all pin layers have been successfully created using the correct gds2 layer and datatype, but other layers have been dropped.

Is there any way to fix or workaround this issue to get the LEF/DEF data for the drawing and boundary layers?
Or do I have to use different purpose names? "pin" seems to be accepted, but not "drawing" nor "boundary".

stefanottili, 2021-06-01:
You're using an OA (OpenAccess) layer map, that's the wrong type of map for lef/def import.
I've not found it in the documentation (yet), but if you look at 
https://github.com/KLayout/klayout/blob/master/src/plugins/streamers/lefdef/db_plugin/dbLEFDEFImporter.cc
at line 113 you find the lef/def purposes listed in static std::string purpose_to_name (LayerPurpose purpose).
This maps the lef/def layer/keywords to layout (gds/oas) layer/datatype pairs.

refob, 2021-06-01:
Your comment helps me a lot!

I took a look at the code and found that klayout uses here the Cadence Encounter streamout layer map format. It would help to document this to tell the people that this is not a DFI or OA stream mapfile.

klayoutmatthias, 2021-06-01:
I don't know Encounter and if I accidentally copied their format it was purely by coincidence.

Some documentation is given in the help of the buddy tools:

```
    --lefdef-map                                 Specifies to use a layer map file

          Use this option to turn off pattern-based layer mapping and to use an
          explicit mapping file instead. See '--lefdef-via-geometry-suffix' for
          a description of the pattern-based mapping scheme.
          
          Using a map file is an alternative way to specify layer mapping. With
          a layer mapping file, the individual target layers need to specified 
          individually for different layer/purpose combinations.
          
          The mapping file is one layer mapping entry per line. Each line is a 
          layer name, followed by a list of purposes (VIA, PIN ...) and a layer
          and datatype number. In addition, 'DIEAREA' can be used to map the 
          design outline to a layer. 'NAME' in place of the layer name and 
          using layer/purpose in the purpose column allows mapping labels to 
          specific layers.
          
          This is an example for a layer map file:
          
          DIEAREA ALL 100 0
          M1 LEFPIN 12 0
          M1 PIN 12 2
          M1 NET 12 3
          M1 SPNET 12 4
          M1 VIA 12 5
          M1 BLOCKAGE 12 10
          NAME M1/PIN 12 10
          VIA1 LEFPIN,VIA,PIN,NET,SPNET 13 0
          M2 LEFPIN,PIN,NET,SPNET,VIA 14 0
          
          If a map file is used, only the layers present in the map file are 
          generated. No other layers are produced. 
```

Matthias

refob, 2021-06-01:
Hallo Matthias,

there is no issue to use this format. GDS2 is also owned by Cadence (via Calma) and you also use it w/o any issues. The same applies to LEF/DEF which was owned by Cadence and was contributed to Si2. In the past there were legal issues for example for Synopsys using the LEF/DEF format. To allow collaboration between all EDA companies and their customers the format is now controlled Si2 which allows all companies that join Si2 to propose enhancements to LEF/DEF. The main development of the LEF/DEF code available at Si2 is still done by Cadence.

Cheers,
Reinhard

klayoutmatthias, 2021-09-24:
I'll close this issue for now. If you want to continue this discussion, please reopen.

Matthias

closed 2021-09-24 by klayoutmatthias