KLayout/klayout #819: Mac DMGs for 0.27.1

Kazzz-S, 2021-06-03:
Dear @klayoutmatthias,

Thank you for making the maintenance release of 0.27.1.

I have branched off 5ccd9990 and updated the build system for Mac to support the Apple Silicon chip.
I could successfully build five different DMGs for Catalina and Big Sur (Intel Mac), respectively.
You can find them in this [Dropbox link (1)](https://www.dropbox.com/sh/sumelo8nr0ibvt6/AAD6ChTqF1TnRYa11dyXmDpea?dl=0).
Some QA tests failed, as you see in these log files in [Dropbox link (2)](https://www.dropbox.com/sh/nq9sr6bzo9zvs2k/AAAXswtlT3cYIeLfumChfwwYa?dl=0).

Could you please investigate the possible causes if you have time?

Regards,
Kazzz-S

klayoutmatthias, 2021-06-03:
Dear @kazzz,

thanks a lot for the DMG's. I have put them on the server.

I have reviewed the QA logs and the major problem still is the source/drain assignment for MOS transistors in netlist extraction. This is depending on STL implementation details as of now and I think that clang's STL is slightly different from the libstdc++ I am using. This causes most of the differences. I am trying some normalization to fix this problem, but it's not in the master branch yet.

There are some other fails which do not worry me - caused by delayed garbage collection on Ruby, some network fails (tlHttp tests - probably caused by the anti-DOS firewall features) and some threading test problem I also encounter upon heavy CPU load.

I'll work on reducing these fails. 

Thanks and best regards,

Matthias

Kazzz-S, 2021-06-23:
I close this ticket since version 0.27.2 has been released.
See #844.

closed 2021-06-23 by Kazzz-S