KLayout/klayout #823: LEF/DEF flow - COMP not supported

refob, 2021-06-04:
Please install the package load_lefdef_tutorial from the Salt.Mine. Then cd to ~/.klayout/salt/load_lefdef_tutorial. Uncomment the COMP line in input/klayout/cmos180.encmap. Then run "make lefdef". You  will see this message:

Warning: Reading layer map file /home/reinhard/.klayout/salt/load_lefdef_tutorial/input/def/../klayout/cmos180.encmap, line 4: COMP entry ignored

In klayout you will see that the DEF boundary and all LEF boundaries are put on the layer "prBoundary.drawing - 100/0". This is only correct for the the DEF boundary.

The LEF boundary boxes should instead go to the layer "instance.drawing 101/0". That is the functionality of the "COMP" line.

stefanottili, 2021-06-04:
a) “comp all” is rightfully not implemented.

It would pointlessly create a rectangle the size of the referenced macro for each and every component reference aka instance aka placement inside the def design layout cell. Since this includes filler cells this would potentially create millions of rectangles. Without any use case for having them that I’m aware of.

Usually the underlying layout cells in gds/oasis matching the macro names have a prBoundary in it. And that’s just for dozens or hundreds of layout cells. “comp all” would just create rectangles overlapping the rectangles already in the referenced cells one hierarchy level below.

b) lef macro to layout cell import correctly uses “diearea all”.

Normally one looks at lef macro geometry in the p&r tool that creates the def.

In the case that one wants to view def with lef macro geometry in klayout and thus convert the macros to layout cells, using the “diearea all” mapping for the prBoundary of the macros matches the behavior of p&r tools that directly write gds/oasis using a non default switch to also write the lef macro geometry.

refob, 2021-06-04:
a) I don't see a point why "comp all" is less or more rightfull then "diearea all". And you misunderstood my request. the data is already there and is translated by klayout. There is BBOX defined in DEF for the top level cell and also BBOXes defined for each leaf cell in LEF. The only this is that "comp all" stores the BBOX in layer 101 instead of layer 100 which is used for the top level BBOX. This will not produce any additonal geometries.

b) I don't understand your point here. LEF is an library input format for the P&R tools which is carefully crafted. The tools should never write out this data to gds or oasis, because too much data is lost. The tools I know write out gds only with the instantiation of the cell and not the leaf cell contents. Later on this data is merged with the library gds/oasis data (which is also golden input data).

stefanottili, 2021-06-04:
A) I have yet to see any flow/tool that needs a rectangle at every cell reference.
But I’ve seen gb gds files due to comp all being enabled (Not so much an issue with oasis).

And if you accept that a rectangle per instance is pointless, then there is.no point in discussing what layer/datatype it should have.

B) if you convert lef macro to layout cells, then that cell should have prBoundary.

refob, 2021-06-05:
a+b) There is no flow where this is required; it is just another debug capability that can be switched on and off. I've seen cells where the prboundary was very different to the calculated boundary of all geometries. I just find it helpfull, if I can put the DEF prboundary on a different layer then the LEF prboundary. Please also take a look at the attached pictures for lefdef and gds.
[image]
[image]

stefanottili, 2021-06-08:
b) For debugging purposes I think that selecting/deleting the def prBoundary since it covers all the lef prBoundaries in the same layer is an acceptable workaround.
Otherwise klayout would have to gain yet another config option.

klayoutmatthias, 2021-06-08:
At least that explains what COMP is about. But I also reason that COMP has an argument (something like a glob pattern?) and consequently it's possible to have multiple COMP statements mapping different component boundaries to different layers, right?

If that's true, that's a rich new feature, not just a config option :(

Matthias

stefanottili, 2021-06-08:
The config option would be to use diearea all for lef prBoundary by default, but optionally use comp all layer datatype for lef prBoundary, since it otherwise is „overlapped“ by the def diearea prBoundary. But I don’t think this is needed at all. 

I’ve never seen the comp used with anything but all. And comp all should never be implemented, it has no usecase that I known of and has been seen to unnecessarily add 100‘s of MB to gds files.

So I would consider this bug to be closed.

klayoutmatthias, 2021-09-24:
I don't know what to implement. I feel that this is a matter of taste.

I'll close this ticket for now. If you want to continue this discussion, please reopen.

Thanks,

Matthias

closed 2021-09-24 by klayoutmatthias