KLayout/klayout #824: LEF/DEF flow - duplicate text label generated

refob, 2021-06-04:
Please install the package load_lefdef_tutorial from the Salt.Mine. Then cd to ~/.klayout/salt/load_lefdef_tutorial. Then run "make lefdef".

In klayout show cell BUFX2 as new top cell. Watch the blue text labels A+Y+vdd+gnd.

Now hide the layer "metal1.pin -81/0". You can see that these labels also exist on layer "text.drawing - 109/0". 

This is not correct. The mapfile input/klayout/cmos180.encmap should put

LEF pins only on layer 109/0
metal1		LEFPIN		81	0

DEF pins only on layers 81, 82 etc.
NAME		metal1/PIN	81	0
NAME		metal2/PIN	82	0

You can also check the toplevel cell - you also find duplicate labels there. There should be nothing on 109/0 according ot the map file.

klayoutmatthias, 2021-06-08:
@refob Thanks for that testcase and the impressive tutorial!

I was not aware that labels are differentiated by purpose. Right now, there is only one "label" purpose and both LEF and DEF pin purposes are combined.

My proposed fix is to supply a new mapping purpose "LEF pin label" which I map "NAME M1/LEFPIN ..." to:

[image]

Matthias

refob, 2021-06-09:
The proposed mapping is fine. I suggest change the labels from "Pins (*)" to "DEF PIns (*)" and from "Pin Labels" to "DEF Pin Labels" to clarify the sources more accurately.

klayoutmatthias, 2021-06-10:
Thanks for this hint. I changed it.

closed 2021-07-06 by klayoutmatthias