KLayout/klayout #832: Segfaults in Wayland

YibaiMeng, 2021-06-10:
When starting klayout 0.26.2 in Ubuntu 21.04 (recently updated to Wayland windowing system), it segfaults with the following output:
```bash
$ klayout
Info: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Aborted (core dumped)
```
Asking it to use X11 solves the problem:
```
XDG_SESSION_TYPE=x11 klayout
```
It seems that Qt 5 [supports wayland](https://wayland.freedesktop.org/toolkits.html). Perhaps a configuration bug?

klayoutmatthias, 2021-06-10:
@YibaiMeng I'm not responsible for the Qt configuration on Ubuntu. I just take the system's Qt5. This is not a KLayout bug.

What happens if you follow the error messages advice and set QT_QPA_PLATFORM=wayland?

Matthias

ChrisZonghaoLi, 2022-10-12:
> It does not work to set the QT_QPA_PLATFORM=wayland. So far the only that works is to launch Klayout like this:
> 
> XDG_SESSION_TYPE=x11 klayout
> 
> It makes a bunch of errors at launch, but amazingly it is working after.

Yes same here, so is it considered as an official workaround to this?

closed 2022-12-06 by klayoutmatthias