KLayout/klayout #838: dbedgeprocess's process function run slow in vs2013

wolfgold, 2021-06-12:
I compiled the contents of TL and DB in 0.27 and 0.25 versions using VS2013.  Remove the GSI part of it,I'm going to use Boolean, merge,size in there, and I'm calling Boolean in dbShapeProcess, which is the way I use a function

```c++
       //m_layout was successfully read from the GDS 
	unsigned int layer_i = m_layout.insert_layer();
	db::LayerProperties tmp_layer;
	tmp_layer.layer = 1000;
	tmp_layer.datatype = 0;
	m_layout.set_properties(layer_i, tmp_layer);

	db::ShapeProcessor sp(true);
	db::BooleanOp::BoolOp op = db::BooleanOp::Xor;

        //layer1 , layer2 is layer index that needed to process
        // top is m_layout's topcell index
	sp.boolean(m_layout, m_layout.cell(top), layer1,
			m_layout, m_layout.cell(top), layer2,
			m_layout.cell(top).shapes(layer_i), op, true, true, true);
	}
```

Even though I used the O2 optimizations in VS, I ran Boolean for exe compiled under release much longer than the Boolean for klayout.I found that the Boolean that consumes the most time is the` process` function in dbedgeprocess.cc.Are there any other factors that affect the efficiency of this function?

klayoutmatthias, 2021-06-14:
@wolfgold  Seriously, you cannot take out code, remove parts of it, compile that with some compiler, with some library, some kind of settings and then expect me somehow magically figure out what's going wrong.

0. This is not a bug. It's your own adventure, so please don't raise a ticket against KLayout.
1. I do not support external use of C++ API. Use on your risk and please be aware that you are legally bound to GPL. So I hope your own project is GPL'd too.
2. I do not support VS2013. I'm using VS2017 and that's the only supported compiler.
3. Performance heavily depends on the STL you're using.
4. There are more options to control performance, specifically STL runtime assertions. Study the compiler options to figure out how.

closed 2021-07-11 by wolfgold

wolfgold, 2022-11-14:
> 
Thank you for you reply,I compiled the program with mingw and it has the normal performance. I think the difference caused by STL.