KLayout/klayout #841: salt security

lukasc-ubc, 2021-06-19:
Hi @klayoutmatthias 

I was trying to update my package (siepic_tools) on salt, and I found that I cannot, even though I have been the owner for many years: 

     A package with the same name already exists ("siepic_tools") and was registered by a different user than yourself. The new package cannot be registered again with the same name. 

     The name is taken from the package definition file at https://github.com/siepic/SiEPIC-Tools.git/tags/v0.3.78/klayout_dot_config/grain.xml (<name> element). 

I tried deleting, and received a helpful comment that it now belongs to someone else!

     The request was successfully created. You will receive a mail to frederick.flynn.15@gmail.com for confirmation soon. 

Do you know how this could happen?  

Could you please return it to my control?

thank you
Lukas

lukasc-ubc, 2021-06-19:
@klayoutmatthias 

I tried to create a different <name> of siepic-tools (instead of siepic_tools) for now, but I receive the following error:

     The package definition file could be downloaded from the specified URL, but the format is not valid. 
     This means that either the XML format is not correct or the name is not valid. 
     Please check whether the name is present (<name> element), not empty and composed of either: letters, digits, "_", "-", "." or "/". 
     This is the link used: https://github.com/siepic/SiEPIC-Tools.git/tags/v0.3.78/klayout_dot_config/grain.xml.

Then I created another release v0.3.78b, with a name of siepic_tools2, and that worked.

When you get a chance, could you please help fix this?  Delete both siepic_tools and siepic_tools2, and create SiEPIC-Tools based on https://github.com/siepic/SiEPIC-Tools.git/tags/v0.3.78c/klayout_dot_config/grain.xml.

I'm really curious how someone was able to hack your system...

thank you very much,
Lukas

klayoutmatthias, 2021-06-19:
Hi Lukas,

I'm sorry this happened. The database wasn't hacked, but somehow someone managed to create database entries with the same package names and those shadowed yours. I deleted the false ones and you should be in control of the original package again. 

I have changed your siepic_tools package to v0.3.78. The siepic_ebeam_pdk package is on v0.3.34. Drop me a not if you'd like to delete the siepic_tools2 package.

Please check if you have access again. 

I'll try to figure out how this can happen. 

Best regards,

Matthias

klayoutmatthias, 2021-06-19:
Hi Lukas,

I have tried to reproduce the problem, but without success.

Someone who tries to register an existing package again with a different mail address will not be able to do so once a package is active. Only between registration and confirmation, someone else could basically re-register a package. But I don't think this was the case here.

I'll check the logs for more hints.

Best regards,

Matthias

klayoutmatthias, 2021-06-19:
I have deleted siepic_tools2 and changed siepic_tools to point to https://github.com/siepic/SiEPIC-Tools.git/tags/v0.3.78c/klayout_dot_config.

I could not rename it to SiEPIC-Tools as unfortunately a dash isn't a valid character for package names - underscores and dots are. I can change that - but only for new KLayout versions which will make the package non-installable for old versions.

I left the name "siepic_tools" which gives an ugly error in KLayout's package manager, but installation works. Maybe you can rename the package to "SiEPIC_tools" in grain.xml. I'll change the database then.

Best regards,

Matthias

lukasc-ubc, 2021-06-20:
Dear @klayoutmatthias 

Thank you.  I have returned the name to "siepic_tools"; it appears that it is case sensitive.  Regarding the "-", the message I received from the salt web page actually included the "-", which is why I used it.  See the 2nd post above.

I don't know if this is the expected behaviour, but I was able to click on the 'update' link the salt package manager sent me for siepic_tools2, and put in the new release with the name "siepic_tools".  So there may be some loop hole?

thank you

klayoutmatthias, 2021-06-20:
Hi Lukas,

you're right, the web server message is not correct. I updated KLayout now to accept "-" as well and I will include that in the next minor release.

I have for now created a test instance of the package repo which can be used for testing: https://sami-test.klayout.org. To enable it, set the "KLAYOUT_SALT_MINE" environment variable to "https://sami-test.klayout.org/repository.xml".

I looking through the logs right now and I'm starting to understand what happened: it seems to be possible to replicate a package by reusing a retired URL. I'll check how I can avoid this problem.

Best regards,

Matthias

klayoutmatthias, 2021-06-20:
Hi Lukas,

After scanning through tons of log entries I think I can tell the story.

* At 2021-06-14T08:10:58 (UTC I assume) you registered a new version of siepic_tools (0.3.77)
* I see that a mail was sent, but I don't see the confirmation happening. So maybe you missed to click on the confirmation link? (no blaming meant, I'm just trying to understand)
* In effect, the package became "WIP" state (updated, but not confirmed). In this state, the package becomes invisible
* I see one attempt by a user to add the package again, but this failed as you still own the package
* After 3 days a clean-up background job deleted the package at 2021-06-18T00:55:03
* This made way for the mentioned user to re-register the package at 2021-06-19T08:58:46 and ownership was transferred

So bottom line: I don't see a hacking attempt - it's rather people trying to help :)

And I also don't see a security hole, except that the mail addresses are being uncovered which is a violation of the promise not to do so. That's to blame on me. And although it helped you: my apologies to frederick.flynn.15.

Does this make sense?

Best regards,

Matthias

lukasc-ubc, 2021-06-21:
Thanks @klayoutmatthias for your deep investigations.

Regarding the confirmation email, there have been cases where the email takes a really long time to arrive.  Perhaps that happened in this case and I forgot to search for it the next day and click on the link.

Based on this experience, I would suggest changing the behaviour of the server.  Namely, rather than making the package invisible during the update time, and then deleting it, instead:
- keep the previous package live until the new one is confirmed
- delete the update if it isn't confirmed

This way there is no accidental removal of the package.  Nor down-time when the package is being updated.

Now I understand the problem that users faced -- the package was missing!

thanks!
Lukas

lukasc-ubc, 2021-09-15:
Hi @klayoutmatthias,

I submitted an update for siepic_tools about 14 hours ago, and haven't received the email.  In the meantime, siepic_tools is missing from the package manager.  Can you please investigate?

thank you
Lukas

klayoutmatthias, 2021-09-16:
I have updated the Salt.Mine application on the server in two respects:
* Packages for which an update is registered remain in place until the update is confirmed (update = new URL, same name)
* It is possible now to re-submit a registration with the same URL (either new or update). A new mail will be sent. Important: for confirmation of the registration, only the *latest* mail's links are valid.

There is a sandbox installation for Salt.Mine to try out the new scheme here: https://sami-test.klayout.org

lukasc-ubc, 2021-11-02:
@klayoutmatthias I just updated two repositories and it worked.  I also checked that the repo is in place during the transition.

thank you!

closed 2022-01-30 by klayoutmatthias

lukasc-ubc, 2022-01-30:
Hi Matthias,

It has been working well for the past several months.  Thank you!