KLayout/klayout #842: ERROR: undefined method `local_variable_get' for #<Binding:0x00000006cd9e60> in Executable::execute

refob, 2021-06-19:
Running DRC I get this error:
<pre>
"space" in: drc_freepdk45.lydrc:314
    Edge pairs: 0 (flat)  0 (hierarchical)
    Elapsed: 0.020s  Memory: 1120.00M
"output" in: drc_freepdk45.lydrc:314
    Edge pairs: 0 (flat)  0 (hierarchical)
    Elapsed: 0.020s  Memory: 1120.00M
GRID section
ERROR: In scr/drc_freepdk45.lydrc: undefined method `local_variable_get' for #<Binding:0x00000006cd9e60>
Writing report database: /home/reinhard/INV_drclvs/drc_report.lyrdb ..
Total elapsed: 11.580s  Memory: 1120.00M
ERROR: undefined method `local_variable_get' for #<Binding:0x00000006cd9e60> in Executable::execute
  scr/drc_freepdk45.lydrc:330:in `block in execute'
  scr/drc_freepdk45.lydrc:328:in `each'
  scr/drc_freepdk45.lydrc:328:in `execute'
  :/built-in-macros/drc_interpreters.lym:27:in `instance_eval'
  :/built-in-macros/drc_interpreters.lym:27:in `execute'
</pre>
Looking at the lines 328-330 of drc_freepdk45.lydr I see no explanation for this error, so I guess that it is a software issue.

You can copy the example data from https://github.com/refob/INV_drclvs. Then "cd INV_drclvs" and run "make drc".

klayoutmatthias, 2021-06-20:
The "Binding#local_variable_get" method has been introduced in Ruby 2.1.0. I assume the version you're using is too old.

What's the platform and version you're using?

Matthias

refob, 2021-06-20:
I'm running CentOS7. The current Ruby version there is:

ruby -v
ruby 2.0.0p648 (2015-12-16) [x86_64-linux]

klayoutmatthias, 2021-06-20:
@refob Ruby 2.0.0 is too old to support "local_variable_get". 

Instead of "binding.local_variable_get(name)" you can also use "eval(name)". This is not precisely the same, but will do the job here I guess.

Matthias

refob, 2021-06-21:
I tried it:
--
grid = 2.5.nm
all_drawing = [ :well, :active, :vtg, :vth, :pplus, :nplus, :poly, 
:thkox, :cont, :metal1, :via1, :metal2, :via2, :metal3, :via3, :metal4, 
:via4, :metal5, :via5, :metal6, :via6, :metal7, :via7, :metal8, :via8, 
:metal9, :via9, :metal10 ]
all_drawing.each do |dwg|
   # a Ruby idiom to get the value of a variable whose name is in "dwg" 
(as symbol)
   #layer = binding.local_variable_get(dwg)
   layer = eval(dwg)
   r_grid = layer.ongrid(grid).polygons(10.nm)
   r_grid.output("GRID: vertexes on layer #{dwg} not on grid of 
#{'%.12g' % grid}")
end
end
--

It returned with:
--
ERROR: In ../scr/drc_freepdk45.lydrc: no implicit conversion of Symbol 
into String
--
Any ideas?

klayoutmatthias, 2021-06-21:
@refob In this case the correct statement is

```
layer = eval(dwg.to_s)
```

Matthias

refob, 2021-06-22:
Work fine - thank you!

closed 2021-08-24 by klayoutmatthias