KLayout/klayout #844: Mac DMGs for 0.27.2

Kazzz-S, 2021-06-23:
Dear @klayoutmatthias,

Thank you for the maintenance release.

As usual, different DMG packages of **KLayout 0.27.2 for Mac** are available as below ([0.27.2 Dropbox Link](https://www.dropbox.com/sh/l6jin4ggqsm4am1/AAD5kLFVG0QhQVTG6GT2MWtwa?dl=0)).
The QA test log files are [here](https://www.dropbox.com/sh/osw5lr01olvxr2a/AACRg1y5E93B25_rEypIq-Nga?dl=0).
**Please note that _pymod_ is not included in those packages.**

## DMG file naming convention ##
**0.27.2 for Catalina** is taken as an example here.
[image]

where
(a) Type of package:  ST-: standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt5 type:  qt5MP: MacPorts; qt5Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:   sys: OS-bundled; mp27: Ruby2.7 from MacPorts; hb27: Ruby2.7 from Homebrew; ana3: Ruby2.5 from Anaconda3
(f) **P**ython type: sys: OS-bundled; mp38: Python 3.8 from MacPorts; hb38: Python 3.8 from Homebrew; ana3: Python3.8 from Anaconda3

## For the primary platform, macOS Catalina (10.15) ##
**LW- and HW- packages use Python 3.8 in MacPorts, Homebrew and Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
1|Catalina (10.15)|ST-klayout-0.27.2-macOS-Catalina-1-qt5MP-RsysPsys.dmg|Homebrew|OS-bundled|OS-bundled|(1)|
2|Catalina (10.15) with MacPorts dev. env.|LW-klayout-0.27.2-macOS-Catalina-1-qt5MP-Rmp27Pmp38.dmg|MacPorts|MacPorts|MacPorts|(2)|
3|Catalina (10.15) with Homebrew dev. env.|LW-klayout-0.27.2-macOS-Catalina-1-qt5Brew-Rhb27Phb38.dmg|Homebrew|Homebrew|Homebrew|(3)|
4|Catalina (10.15) with Anaconda3 dev. env.|LW-klayout-0.27.2-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(4)|
5|Catalina (10.15)|HW-klayout-0.27.2-macOS-Catalina-1-qt5Brew-RsysPhb38.dmg|Homebrew|OS-bundled|Homebrew|(5)|

(1) shares OS-bundled Ruby2.6 and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).
(2) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(3) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(4) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(5) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.8 from Homebrew are embedded, hence, heavy.

## For the latest macOS Big Sur (11.4) ##
Five macOS DMGs for **Big Sur** are ready for experimental purposes.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 12.1.2.

**LW- and HW- packages use Python 3.8 in MacPorts, Homebrew and Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
21|BigSur (11.4)|ST-klayout-0.27.2-macOS-BigSur-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(21)|
22|BigSur (11.4) with MacPorts dev. env.|LW-klayout-0.27.2-macOS-BigSur-1-qt5MP-Rmp27Pmp38.dmg|MacPorts|MacPorts|MacPorts|(22)|
23|BigSur (11.4) with Homebrew dev. env.|LW-klayout-0.27.2-macOS-BigSur-1-qt5Brew-Rhb27Phb38.dmg|Homebrew|Homebrew|Homebrew|(23)|
24|BigSur (11.4) with Anaconda3 dev. env.|LW-klayout-0.27.2-macOS-BigSur-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(24)|
25|BigSur (11.4)|HW-klayout-0.27.2-macOS-BigSur-1-qt5Brew-RsysPhb38.dmg|Homebrew|OS-bundled|Homebrew|(25)|

(21) **experimental**; shares OS-bundled Ruby2.6 and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).
(22) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(23) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(24) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(25) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.8 from Homebrew are embedded, hence, heavy.

## For previous OS ##
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Mojave (10.14)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
7|HighSierra (10.13)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
8|Sierra (10.12)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
9|ElCapitan (10.11)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
x|Yosemite (10.10)|has been dropped due to the compiler issue reported in #360.|

(1p) I expect that build is still possible.

Please close this ticket after publishing the packages.
With warm regards,
Kazzz-S

klayoutmatthias, 2021-06-23:
:) 

Many thanks! I uploaded the files.

I also was happy to see the tests are basically clean. There is a small (known) issue with the random nature of the Ruby GC, but that's it.

Thanks and best regards,

Matthias

closed 2021-06-23 by klayoutmatthias

Kazzz-S, 2021-07-14:
Dear @klayoutmatthias,

I have uploaded **the second DMG** for `Big Sur + Anaconda3` to address [this issue.](https://www.klayout.de/forum/discussion/1887/anaconda3-install-on-big-sur)
* LW-klayout-0.27.2-macOS-BigSur-2-qt5Ana3-Rana3Pana3.dmg
* LW-klayout-0.27.2-macOS-BigSur-2-qt5Ana3-Rana3Pana3.dmg.md5

They are stored in this [Dropbox link](https://www.dropbox.com/sh/l6jin4ggqsm4am1/AAD5kLFVG0QhQVTG6GT2MWtwa?dl=0) as usual.
Please update the [download site](https://www.klayout.de/build.html) by replacing the existing one with the new one,
then add a comment to [the post.](https://www.klayout.de/forum/discussion/1887/anaconda3-install-on-big-sur)

With warm regards,
Kazzz-S