KLayout/klayout #848: Too short progress bar

Kazzz-S, 2021-06-28:
The progress bar looks too short :-)
KLayout Version: 0.27.2
Build: Source build on Linux (Mac, too)

[image]

Thanks,
Kazzz-S

klayoutmatthias, 2021-06-28:
Well, it was very long before ... :)

Actually, there are sub-progress bars too for nested operations. Previously that was stacked, but that created ugly flickers in the UI. I now arranged them horizontally to have them occupy the same height. The sub-progress bars are attached to the left of the short one.

That was a low hanging fruit to achieve a less annoying progress bar.

Kazzz-S, 2021-06-28:
I was unaware of this change until yesterday.
Thank you for your explanation.

Kazzz-S

closed 2021-06-28 by Kazzz-S