KLayout/klayout #860: Library registration

lukasc-ubc, 2021-07-02:
Hi Matthias,

for Library.register, the documentation states that it replaces the library. 

https://www.klayout.de/doc-qt5/code/class_Library.html#method24

self.register(library)

However, if I run the library multiple times, it shows up repeated:

[image]

In 0.26.12, it replaced correctly.  Furthermore, documentation says that this and other features were introduced in 0.27, but they were already in 0.26.

I checked this in version 0.27.2, klayout HW-klayout-0.27.2-macOS-BigSur-1-qt5Brew-RsysPhb38

klayoutmatthias, 2021-07-06:
This is the same as in #810, right?

So I copy my attempt to reproduce it here:

------------------
Hi Lucas,

I tried first to reproduce the "library.register" problem, but I can't confirm that (Linux).

This is what works for me:

```
lib_a = RBA::Library::new
lib_a.layout.add_cell("LIB_A")
lib_a.register("ISSUE_810_LIB")

lib_b = RBA::Library::new
lib_b.layout.add_cell("LIB_B")
lib_b.add_technology("si4all")
lib_b.register("ISSUE_810_LIB")

# register another library for tech "si4all" which replaces lib_b
lib_c = RBA::Library::new
lib_c.layout.add_cell("LIB_C")
lib_c.add_technology("si4all")
lib_c.register("ISSUE_810_LIB")

# register another library for "no tech" which replaces lib_a
lib_x = RBA::Library::new
lib_x.layout.add_cell("LIB_X")
lib_x.register("ISSUE_810_LIB")
```

This will correctly register two libraries "ISSUE_810_LIB", one for "si4all" technology, the other unspecific. The first two "register" calls are replaced.

JinWallner, 2021-07-26:
Hi Lucas,

I see the same issue. After I move the self.register(library) to the end after the self.technology=tech_name or self.add_technology(tech_name), the issue is gone.