KLayout/klayout #863: Reading extra cells from CLI?

donn, 2021-07-12:
I'm trying to view a def file in Klayout from the CLI (to avoid the whole File > Open song and dance every time). I can't find an option that's equivalent to this:

[image]

Any ideas?

klayoutmatthias, 2021-07-14:
@donn You can probably do that with a script. But what exactly is the plan? Importing LEF files into an existing file? Problem with LEF is that usually these macros do not contain much information. And there are manifold configuration options for LEF (see stream reader options).

donn, 2021-07-15:
Yes, precisely that. I import a hierarchical def file made of cells and want to import cells from another lef file (it indicates the cells not found otherwise, even if the lef file is imported in the "with lef files" option.) In any case, thank you for confirming this cannot be done purely via the CLI.

closed 2021-07-15 by donn