KLayout/klayout #866: Trace All Nets->Hierarchical->ProbeNet doesn't work in 0.27.2

JinWallner, 2021-07-15:
Hi, I recently switched to 0.27.2. I can't probe net with Trace All Nets->Hierarchical->ProbeNet anymore. The Flat one works but not the Hierarchical one.

klayoutmatthias, 2021-07-17:
Sorry, I can't confirm than.

Test layout is this: https://github.com/KLayout/klayout/blob/master/testdata/gds/t10.gds

Layer stack is this:
[image]

I use Tools -> Trace All Nets -> Hierarchical.

I click on "Probe" and on a relevant shape in the layout. The net gets highlighted and selected in the net browser.

OS is Ubuntu 20.

Matthias

klayoutmatthias, 2021-07-17:
I just confirmed the issue cannot reproduced either on Windows 10, 64bit.

JinWallner, 2021-07-19:
> Sorry, I can't confirm that.

Hi Matthias,

Thank you for your response. I am on Windows10. It's very strange that I
can use the Hierarchical Probe on your test layout. But I tried to make a
very simple random layout with similar layers. And the probe only works
within the cell. I just can't get it to work at the Top cell. It only
highlights the whole cell instance. What do I miss here?

Thanks,

Jin

klayoutmatthias, 2021-07-19:
Hi Jin,

maybe it's possible to share the layout and the layer stack you're using? I'd like to investigate the issue.

Matthias

JinWallner, 2021-07-19:
Hi Matthias,

Here is my test layout.
https://github.com/JinWallner/public/blob/master/test.gds
I drew random shapes with layer 3, 5, 6, and I defined the same
connectivity as your example. But somehow the net race only works in the
cell "a" but not the cell "Top".

Thanks,

Jin

klayoutmatthias, 2021-07-19:
Hi Jin,

thanks for the sample. I see the problem.

The reason is actually that the "a" subcell does not have a pin leading to the TOP cell. This means that although instantiated in the layout, the "a" cell is not a subcircuit of "TOP" in the netlist. So eventually the layout and netlist hierarchy does not match.

I'll see if I can rectify the problem. This will need a special kind of subcircuit without a pin.

Thanks and best regards,

Matthias

commit 585baa46 (2021-07-21): Fixed #866 (probe net not working for floating subcircuits in hierarchical net tracer)
commit ce6f63b2 (2021-07-21): Merge pull request #873 from KLayout/issue-866

Fixed #866 (probe net not working for floating subcircuits in hierarc…
closed 2021-07-21 by klayoutmatthias