KLayout/klayout #867: Macro Development issue in 0.27.2

JinWallner, 2021-07-15:
The search function in the macro development in 0.27.2 doesn't seem to work.  It still highlights the words you want to search. But clicking the magnifying glass icon won't bring you to the next matched word. And the Macro Development window seems to freeze after that. Can't do any edit after that.

klayoutmatthias, 2021-07-17:
Sorry, can't confirm that either.

You're describing bugs which have been there in 0.27.1 or earlier. Are you sure you see 0.27.2? 

And what's the OS you're on?

klayoutmatthias, 2021-07-17:
Here is a video demonstrating how it works for me:

https://user-images.githubusercontent.com/25720841/126036565-1531c17b-4b07-4a0c-abf7-fe7e9c95d3e2.mp4

JinWallner, 2021-07-19:
Hi Mattias,

Thank you very much. I did more testing. It seems that issue is associated
with the SiEPIC package I am using. After removing the package, the macro
development enviroment works fine. It seems the package has a compatibility
issue with 0.27.2.

Thanks,

Jin

klayoutmatthias, 2021-07-19:
Hi Jin,

it's hard to believe a package has such an effect. But I'll try to reproduce the problem with this package installed.

Matthias

JinWallner, 2021-07-19:
You are right. The problem came back even after removing all the packages.
I was able to repeat the issue now. If I open a file and start a search
without clicking the tab first, it will trigger the issue. If I click the
tab, then everything is fine.

klayoutmatthias, 2021-07-19:
Very good. That detail was missing.

I can now reproduce the problem and fix it.

Thanks and best regards,

Matthias

lukasc-ubc, 2021-07-19:
Great. Glad it isn’t a SiEPIC package problem!

klayoutmatthias, 2021-07-19:
Yes, I doubted so :)

commit 2570c146 (2021-07-21): Merge pull request #872 from KLayout/issue-867

Fixed #867 (macro editor freeze after search next)
commit 52f7a048 (2021-07-21): Fixed #867 (macro editor freeze after search next)
closed 2021-07-21 by klayoutmatthias