KLayout/klayout #871: klayout 0.27.2 crashes on simple gds with AREF

patrickschulz, 2021-07-19:
Hi,
I using a layout generator that creates GDSII files. I never had an issue with klayout before, but now it crashes/freezes on a very simple gds. I've narrowed the test case down to a cell with an 3 x 3 AREF. If the layer (0/0) is not displayed at the beginning, everything works fine. I can also display the layer when I descend into the hierarchy (show only the subcell). However, going back or showing the layer in the topcell, the crash occurs. Then, klayout does not react to anything and I have to kill it.

The problem seems to occur with different layer properties/different technology settings. More oddly, an AREF of, say, 2 x 3 works just fine. I'm on archlinux, using klayout version 0.27.2.

The text representation of the gds is (I also attached the binary file):

      HEADER #(   6): { 258 }
      BGNLIB #(  28): { 2021 7 19 9 53 55 2021 7 19 9 53 55 }
     LIBNAME #(  12): { testlib }
       UNITS #(  20): { 1.0 1e-09 }
      BGNSTR #(  28): { 2021 7 19 9 53 55 2021 7 19 9 53 55 }
     STRNAME #(  16): { groundmesh_1 }
    BOUNDARY #(   4)
       LAYER #(   6): { 0 }
    DATATYPE #(   6): { 0 }
          XY #(  44): { -100 -100 100 -100 100 100 -100 100 -100 -100 }
       ENDEL #(   4)
      ENDSTR #(   4)
      BGNSTR #(  28): { 2021 7 19 9 53 55 2021 7 19 9 53 55 }
     STRNAME #(  12): { toplevel }
        AREF #(   4)
       SNAME #(  16): { groundmesh_1 }
      COLROW #(   8): { 3 3 }
          XY #(  28): { 0 0 600 0 0 600 }
       ENDEL #(   4)
      ENDSTR #(   4)
      ENDLIB #(   4)

[test.zip](https://github.com/KLayout/klayout/files/6838949/test.zip)

Thanks, kind regards,
Patrick

klayoutmatthias, 2021-07-19:
Hi Patrick,

that is a super simple GDS and should display without issues. It's a perfectly normal GDS and I'm using such GDS files very often and without problems.

I'm afraid I cannot see this issue on my Ubuntu 20 Linux Box:

https://user-images.githubusercontent.com/25720841/126204069-5330603a-6f94-4bf5-8f3c-8b17f3cbb6fa.mp4

Matthias

patrickschulz, 2021-07-19:
Hi Matthias,
thanks for your fast reply. My OS is archlinux. A clean .klayout folder helped me narrow down the issue. I can reproduce the bug with a klayoutrc like this:

    <?xml version="1.0" encoding="utf-8"?>
    <config>
      <draw-array-border-instances>true</draw-array-border-instances>
      <full-hierarchy-new-cell>true</full-hierarchy-new-cell>
    </config>

But regardless of the value of full-hiearchy-new-cell, when showing the full hierarchy also of an already opened cell, klayout freezes. With draw-array-border-instances set to true, I can't show the full hierarchy. 

Does that help? I can also make a video, if that is helpful. 

What exactly does draw-array-border-instances do, by the way? I think I enabled it rather accidentally and never had a problem with it. It seems that it somehow (at least for me) has an issue with full hierarchy and a special AREF. Weird.

Kind regards,
Patrick

klayoutmatthias, 2021-07-20:
The questionable option is here:

[image]

and is found here:

[image]

the effect is that:

[image]

I don't think it's really useful but it was requested by users many years ago.

Matthias

commit 92419801 (2021-07-21): Fixed #871 (application stalls with 'draw only border instances' mode with AREFs)
commit b7f4c404 (2021-07-21): Merge pull request #874 from KLayout/issue-871

Fixed #871 (application stalls with 'draw only border instances' mode…
closed 2021-07-21 by klayoutmatthias