KLayout/klayout #883: Net tracing doesn't work for pcells

JinWallner, 2021-07-30:
Hi,

For version 0.27.3, the net tracing function doesn't work for pcells. It only works after I save the layout without the pcell and library information. Is that expected behavior?

Thanks,

Jin

klayoutmatthias, 2021-07-30:
Hi Jin,

Here is a forum post related to PCells and Net tracer: https://www.klayout.de/forum/discussion/comment/5908#Comment_5908

The net tracer traces "through" PCells, but not "at" PCells. "Trace All Nets" however always works.

Matthias

JinWallner, 2021-08-02:
Hi Matthias,

Thanks. I noticed that the limitation is not just for PCells. It doesn't work even for a static cell that is associated with a library. It works fine after I save the layout without the library information. 

Thanks,

Jin

klayoutmatthias, 2021-08-04:
Yes, that is the effect discussed in the forum post. 

Shapes living inside library members or PCells are not considered selectable as they cannot be edited (a library or PCell is only selectable as a whole). Hence clicking on such a shape does not produce the shape and therefore tracing will not know where to start.

Matthias

JinWallner, 2021-08-04:
Thank you for the explanation. 

Jin

closed 2021-08-04 by JinWallner