KLayout/klayout #885: Allowing mismatch of sub-cells

mguthaus, 2021-08-04:
Many designs will have "must connect" pins in a layout that are disconnected at some level of hierarchy but connected later. One such example might be a supply pin. Often the schematic will show them mismatching at a lower level of hierarchy, but then the higher level of hierarchy passes after they are connected. It seems that the algorithm used in klayout precludes this. From the web site:

https://www.klayout.de/doc-qt5/manual/lvs_compare.html

it will fail to match if any subcell mismatches. 

Yes, you could specify that they are connected (using connect_explicit, for example) in the LVS script to avoid the problem, but then this might actually mask an error where they are disconnected at the top level. 

Magic, for example, will flatten a mismatching subcell and try to match the next higher level. I believe that this is the correct behavior.

klayoutmatthias, 2021-08-04:
Hi Mathew,

I think I see the point.

You can basically have flattening already when you use 'flatten("CIRCUIT")' on the extracted and schematic netlist.

But I wonder if flattening is helpful - I think that this will make debugging much more difficult and after all, the netlists and hence compare times will increase.

So I wonder whether there are smarter solutions, such as using "connect_explicit" or "connect_implicit" and adding a check which ensures such nets are fully connected.

Matthias

mguthaus, 2021-08-04:
Flattening isn't an option for large designs... the run-time after flattening would be impractical for a chip-level design (or even an IP level design).

I'm not sure that it is possible to easily identify which pins would be must connect. Sometimes it may be a disconnected well inside an instance, sometimes it could be a gate input, often it is a supply as I mentioned, etc. It would be really unfortunate to make it a manual step in LVS since commercial and other open-source tools don't do that. 

Does layout require a 1-to-1 mapping of hierarchy? I have some extra hierarchy layers but they don't have active devices and seem to be removed, but in the general case is it required to have 1-to-1 mapping? It is very common for layout hierarchy to be different than a schematic hierarchy.

closed 2023-11-22 by klayoutmatthias