KLayout/klayout #887: Cell names in round brackets

Lilipos, 2021-08-09:
Hi, 
In the latest klayout versions I noticed problem with cell names. When I open gds file there is several top cells, while should be only one. In hierarchy list there are "new" cell names in round brackets which are empty inside (there should be cell names without those brackets and actually present in the top cells list). There was no such problem in 26.8.0 version and this problem occurred only for some selected gds files.

[image]

Cheers,
Lili

klayoutmatthias, 2021-08-09:
@Lilipos Something is wrong with your GDS files I think. Cells in round brackets indicate cells which are instantiated, but no present in your file ("ghost cells"). From the screenshot it looks a subcell1 is present, still it's not referenced. 

The problem may be due to an invisible character in the cell names, e.g. a blank after "subcell1". Standard GDS files will never look this way.

Is it possible to send such a GDS file for analysis?

Matthias

Lilipos, 2021-08-18:
Hi Matthias, 

Thank you for your response. 
This gds file I received externally to analysis. I tried to modified it for you, but I can't save it as I got Error: "The following cell name(s) are used for more than one cell - can't write this layout: ...".
Is there a way to check what is the difference between cells references in klayout latest and 26.8.0 version when reads gds file?
I can't sent you the gds file in the actual form. 

Lili

klayoutmatthias, 2021-08-18:
Hi Lili, 

I'm sorry, but without a test file I can't do debugging or analysis. It's very difficult for me base this discussion on guessing and frankly I think it's a fair deal to ask for test data in return for a free tool. I can offer that you send me the file over a private channel and I will treat it confidently. Saving the file with KLayout probably is not a good idea as it is already broken after reading.

Nevertheless, I tried to reproduce the problem by faking a GDS file with duplicate cells as the error message suggests. I tried other ways too, but so far failed to reproduce the issue. There was a reconfiguration of the reader due to a more elaborate handling of cell merging, but GDS files are fairly simple things, so I wonder how it is possible to screw up the reader. I still think that well-formed GDS files should not create such problems, but I would like to debug this issue.

Matthias

klayoutmatthias, 2021-08-18:
Update: I was able to break a GDS file in a way that reproduces your problem exactly. Call me a genius for finding out so much with so little information :)

Here is what I did:

A GDS file is made from records. Cells written with a BGNSTR record and are referenced using a SREF record. For BGNSTR records, a following STRNAME record will indicate the cell name and for a SREF record, a following SNAME record will give the cell name.

Here is parts of a dump for a sane GDS file:

```
...
# Definition of a cell called "BBB":
000000062   00 1c 05 02              [BGNSTR]
000000066   07 e5 00 08 00 12 00 15    2021-08-18 21:56:32
000000074 + 00 38 00 20 
000000078   07 e5 00 08 00 12 00 15    2021-08-18 21:56:32
000000086 + 00 38 00 20 
000000090   00 08 06 06              [STRNAME]
000000094   42 42 42 00                "BBB"
...
# Instance of a cell called "BBB":
000000364   00 04 0a 00              [SREF]
000000368   00 08 12 06              [SNAME]
000000372   42 42 42 00                "BBB"
...
```

I patched the file this way:

```
...
000000062   00 1c 05 02              [BGNSTR]
000000066   07 e5 00 08 00 12 00 15    2021-08-18 21:56:32
000000074 + 00 38 00 20 
000000078   07 e5 00 08 00 12 00 15    2021-08-18 21:56:32
000000086 + 00 38 00 20 
000000090   00 08 06 06              [STRNAME]
000000094   42 42 00 00                "BB"
...
000000364   00 04 0a 00              [SREF]
000000368   00 08 12 06              [SNAME]
000000372   42 42 00 00                "BB"
...
```

So essentially the cell names have been turned into "BB" (instead of "BBB"), but in a wrong way. The SNAME record can hold an even number of characters, so "BBB" has to be represented by three "B" and one padding character (0). For "BB" the record should be shorter and two characters are sufficient. No padding needed. In my case I kept the record longer (4 bytes) and included two 0 padding characters.

So the effect is exactly what you observe:

[image]

Even saving isn't possible because KLayout says "BB" is a duplicate cell.

So my conclusion is that you GDS file is in fact broken. Still the reader could be kind of smarter and read such a case correctly. The reader's reconfiguration may have had the side effect to not reading this case the same way than before, but my test suite does not include many ill-formed GDS file. I can update the reader so it will read these files again, but I'd suggest you talk the to supplier of this specific GDS file. They may want to fix their flow because other tools may not like these kind of files either.

Matthias

stefanottili, 2021-08-18:
Hi Matthias,

providing a testcase should always be a requirement.

Even though I agree with your assessment that the gds is broken, it’s still a bug in the reader. 
If you add a 0x00 after any (even) length string and then treat them as a 0 terminated c string, the buggy gds will have been read even if the strings would have different length of 2, 3 or 4 for either strname or sname.

Somehow you must be treating the strname string differently from the sname string, because currently searching the sname in the list of strname to resolve the reference fails.

Stefan

Lilipos, 2021-08-19:
Hi Matthias, 

I just read your messages. Indeed you are a genius :) Sorry if you wasted a lot of time on this, because I didn't send you more information. I'm glad that you find the problem and Klayout is amazing tool. 
I'll contact my supplier and inform about this problem. 
Many thanks! 

Lili

closed 2021-08-19 by Lilipos

klayoutmatthias, 2021-08-23:
@stefanottili is right - although this is a border case I'm going to restore the original bahaviour. I have created a follow-up ticket with a test case: https://github.com/KLayout/klayout/issues/893