KLayout/klayout #888: Klayout SIGSEV error on WSL1/Ubuntu

gribeill, 2021-08-09:
Hi all!

I've been running into repeated SIGSEV errors in when running klayout. Stacktrace is [here](https://gist.github.com/gribeill/d9d294db39ab38ab1fbab05de24b2e19). 

Unfortunately I have not been able to find a minimal test case that reproduces this crash; from my perspective it seems to happen "randomly" while editing a layout, with perhaps a tendency to occur more frequently when using python-scripted pcells. I'd be happy for any suggestions to more effectively debug! 

This is running a built-from-source klayout on WSL1/Ubuntu:
```bash
$ uname -a
Linux [redacted] 4.4.0-18362-Microsoft #1049-Microsoft Thu Aug 14 12:01:00 PST 2020 x86_64 x86_64 x86_64 GNU/Linux
$ klayout -v
KLayout 0.27.4
```

This isn't a critical issue for me as I can run klayout in Windows, but I prefer to run in WSL to allow interfacing with some Linux-only tooling; anything I can do to help figure out what's going on please let me know and I am happy to contribute!

klayoutmatthias, 2021-08-09:
@gribeill I have stopped using WSL myself some time ago, after some frustrating experiences. But that's not necessarily a problem of WSL. 

First of all, you may want to use a released tag, not the master - if you use v0.27.3 you should get get (stable) code base of the 0.27.3 release.

BTW: What is the graphics frontend you're using?

The stack trace isn't telling much I'm afraid. A debug build might give a slightly better stack trace, but the trace already hints towards a problem with the cell tree model which is very tedious to debug. A valgrind log is usually a good way to reveal the root cause of such issues. You can run KLayout within valgrind, but beware that the application is very slow during these sessions.

Matthias

gribeill, 2021-08-09:
@klayoutmatthias thanks for the feedback -- I will build using the release tag and give valgrind a try...

klayoutmatthias, 2021-08-09:
Very good. Thanks :)

I don't expect much differences from the 0.27.3 build, but I cannot rule out that you picked a temporarily unstable version.

Matthias

gribeill, 2021-08-10:
Building with 0.27.3 and Qt5 instead of Qt4 seems to have fixed the crash -- I will fix the issue and reopen if I can reproduce with this build. Thanks again!

closed 2021-08-10 by gribeill