KLayout/klayout #892: Standalone Package is missing "Basic" library

jparra-qorvo, 2021-08-13:
Hello,

With the Standalone version of python (version 0.27.3), I'm attempting to create a gds file containing polygons with custom fonts. In the UI version of the app, I'm able to have everything work if I import it as a macro but unfortunately, we are working in different environments so the standalone package is the best choice. Is there any way to include the Basic library in the standalone module while also including custom fonts (in the form of a gds file)? Or is that feature not feasible at the moment?

jparra-qorvo, 2021-08-13:
Just figured it out. Had to import the klayout.lib to locate the library. But if you can, I would like to know how to add custom fonts to the "Basic" library through Klayout's standalone package.

klayoutmatthias, 2021-08-18:
@jparra-qorvo  Yes, loading "klayout.lib" is required in order to activate the Basic library.

Regarding the custom font I'm afraid there is no way currently to install custom fonts. But it's fairly easy to provide a Python binding for the function that needs to be called in order to set the font search path. This will allow pointing the TEXT cell (and the TextGenerator class) to the font installation directories.

BTW: in order to generate text fields you do not necessarily need a PCell. You can also use the TextGenerator class (https://www.klayout.de/doc-qt5/code/class_TextGenerator.html#k_1) which returns a Region object you can put into your layout.

Matthias

klayoutmatthias, 2021-08-23:
I'll add two new functions for the TextGenerator class. These allow getting and setting the font search paths.

In my setup I have a folder called "/home/matthias/klayout-fonts" with a font file called "new_font.gds". This is how I can use the new font path setter to use this custom font:

```
import klayout.db as db
import klayout.lib as lib

font_paths = [ "/home/matthias/klayout-fonts" ]
db.TextGenerator.set_font_paths(font_paths)

lib = db.Library.library_by_name("Basic")
pcell_id = lib.layout().pcell_id("TEXT")

ly = db.Layout()

# "new_font" corresponds to "new_font.gds"
param = { "layer": db.LayerInfo(1, 0), "text": "ABC", "font_name": "new_font" }
pc = ly.add_pcell_variant(lib, pcell_id, param)

top = ly.create_cell("TOP")
top.insert(db.CellInstArray(pc, db.Trans()))

options = db.SaveLayoutOptions()
options.write_context_info = False
ly.write("output.gds", options)
```

klayoutmatthias, 2021-09-21:
Closing as this is fixed in the master branch.

closed 2021-09-21 by klayoutmatthias