KLayout/klayout #902: Issue with record ordering when using PATHTYPE==4 and BGNEXTN and ENDEXTN

patrickschulz, 2021-09-01:
When using path with path types == 4 (flexible start and end extension), the ordering of the individual records for this path entry matter, in certain cases the parser throws an error. While this works:

    HEADER (6) -> data: 258
    BGNLIB (28) -> data: { 2021 9 1 15 27 10 2021 9 1 15 27 10 }
        LIBNAME (10) -> data: testlib
        UNITS (20) -> data: { 1.0 1e-09 }
        BGNSTR (28) -> data: { 2021 9 1 15 27 10 2021 9 1 15 27 10 }
            STRNAME (16) -> data: opctoplevel
            PATH (4)
                LAYER (6) -> data: 15
                DATATYPE (6) -> data: 0
                PATHTYPE (6) -> data: 4
                WIDTH (8) -> data: 100
                BGNEXTN (8) -> data: 100
                ENDEXTN (8) -> data: 10
                XY (20) -> data: { 0 0 1000 0 }
            ENDEL (4)
        ENDSTR (4)
    ENDLIB (4)

(where WIDTH comes before BGNEXTN and ENDEXTN)

this does not:

    HEADER (6) -> data: 258
    BGNLIB (28) -> data: { 2021 9 1 15 27 10 2021 9 1 15 27 10 }
        LIBNAME (10) -> data: testlib
        UNITS (20) -> data: { 1.0 1e-09 }
        BGNSTR (28) -> data: { 2021 9 1 15 27 10 2021 9 1 15 27 10 }
            STRNAME (16) -> data: opctoplevel
            PATH (4)
                LAYER (6) -> data: 15
                DATATYPE (6) -> data: 0
                PATHTYPE (6) -> data: 4
                BGNEXTN (8) -> data: 100
                ENDEXTN (8) -> data: 10
                WIDTH (8) -> data: 100
                XY (20) -> data: { 0 0 1000 0 }
            ENDEL (4)
        ENDSTR (4)
    ENDLIB (4)

I'm not sure if this is really a bug, since the GDS manual I have on my computer also puts these records in that order (<path> ::= PATH [ELFLAGS] [PLEX] LAYER DATATYPE [PATHTYPE] [WIDTH] [BGNEXTN] [ENDEXTN]). I just stumbled upon this because I'm building a layout generator, where the code is slightly more complicated if it has to respect this, so I did not start with it and so this issue came up. 
I've attached two GDS files showing the issue.
[testcase.zip](https://github.com/KLayout/klayout/files/7091476/testcase.zip)

klayoutmatthias, 2021-09-01:
@patrickschulz Thanks for bringing this up. However, I think that the GDS record order is actually fixed - although in a few cases I allow some reordering (MAG, ANGLE, STRANS for SREF/AREF for example). But so far, I have not seen GDS files which deviate from the predefined order in PATHs.

The question is probably how other tools will read such GDS files. I assume that most GDS consumers expect the records in the specified order. So if you want to stay compatible with the rest of the EDA domain, you should not only care about KLayout.

Matthias

patrickschulz, 2021-09-02:
@klayoutmatthias Thanks, that sound reasonable :)

closed 2022-01-30 by klayoutmatthias