KLayout/klayout #903: RuntimeError: std::bad_alloc in Region.^

ahmednofal, 2021-09-01:
Hello @klayoutmatthias thank you for your work on this great project. 

- Klayout crashes with `RuntimeError: std::bad_alloc in Region.^`  error while perfroming an xor on a certain layer (`95,20` to be exact) between two cells objects which are `sky130_sram_1kbyte_1rw1r_32x256_8`  cells which can be found [here](https://github.com/efabless/sky130_sram_macros/blob/main/sky130_sram_1kbyte_1rw1r_32x256_8/sky130_sram_1kbyte_1rw1r_32x256_8.gds) 
     -  It crashes on this line`             violation_region.insert(pya.Region(cella_layer) ^ pya.Region(cellb_layer))`
     -  When it crashes I still have 4.5 GB of RAM remaining in my system. So it is not `excessive RAM usage issue`.
-  I tried both hierarchical and flat in the xor tool and it does not crash, hierarchical analysis mode is just slower 

- I have also tried to imitate [the xor tool](https://github.dev/KLayout/klayout/blob/master/src/plugins/tools/xor/lay_plugin/layXORToolDialog.cc) drc functionality but I am still reading through the code which can be found [here](https://github.dev/KLayout/klayout/blob/master/src/plugins/tools/xor/lay_plugin/layXORToolDialog.cc), I am trying to imitate it because inside klayout application I can do the xor without crashing through the xor tool.

My code to do xor on two [Cell objects](https://www.klayout.de/doc/code/class_Cell.html)
```
def cell_xor(cella, cellb, flat):
    violation_region = pya.Region()
    # To have all possible layers between the two cells
    for layer_info in set([*cella.layout().layer_infos(), *cellb.layout().layer_infos()]):
        cella_layer = cella.begin_shapes_rec(cella.layout().layer(layer_info.layer, layer_info.datatype))
        cellb_layer = cellb.begin_shapes_rec(cellb.layout().layer(layer_info.layer, layer_info.datatype))
        if not flat:
            violation_region.insert(pya.Region(cella_layer, global_deepshapesstore) ^ pya.Region(cellb_layer, global_deepshapesstore))
        else:
            violation_region.insert(pya.Region(cella_layer) ^ pya.Region(cellb_layer))
        del cella_layer
        del cellb_layer
        gc.collect()
    return violation_region
```

klayoutmatthias, 2021-09-01:
BTW: you don't need to recode the functionality when you need the XOR feature: there is a binary called "strmxor" which wraps the XOR tool into an executable.

klayoutmatthias, 2021-09-24:
Any feedback? I'll close the ticket otherwise.

Matthias

closed 2022-01-30 by klayoutmatthias