KLayout/klayout #904: Importing Lower Hierarchy Cells From LEF Files?

donn, 2021-09-08:
Klayout Version: KLayout 0.27.3
OS: Manjaro GNU/Linux

---
There are multiple aspects to this issue:

### LEF files in the same folder as the same DEF file appear to be loaded automatically, making it so adding something as an 'extra lef file' cause a crash.

Example: with this file structure:

* `merged.lef`
* `open.py`
* `RAM32.placed.def`
* `sky130A.lyt`

This following script fails with `Duplicate MACRO name: sky130_fd_sc_hd__a2bb2o_1 (line=792, cell=, file=merged.lef) in MainWindow.load_layout`. (sky130_fd_sc_hd__a2bb2o_1 is only defined in merged.lef.)

```python
import pya

app = pya.Application.instance()

try:
    win = app.main_window()

    tech = pya.Technology()
    tech.load("./sky130A.lyt")

    layout_options = tech.load_layout_options
    layout_options.lefdef_config.lef_files = ["./merged.lef"]

    cell_view = win.load_layout("./RAM32.placed.def", layout_options, 0)
except Exception as e:
    print(e)
    app.exit(-1)
```

I've attached a reproducible for your convenience: [issue_a.tar.gz](https://github.com/KLayout/klayout/files/7125438/issue_a.tar.gz)

Workaround: remove merged.lef from the lef_files as part of lefdef_config, however if there are unrelated lef files in the same folder that may have the same macro name, they have to be removed as well.

---

B. Cells loaded from lef files need to be re-imported as extra cells to be shown in hierarchy

This DEF file is made up of cells. Unfortunately, these cells do not show up, and appear to be missing (as evident by the `(` `)` surrounding them). This is the minimum hierarchy that you can see after loading the .def file:

[image]

The only way to see the cells is to re-import the .lef file as follows:

[image]

Following this step, you can see the cells as you'd normally expect:

[image]

Permit me to say that this step is not really an obvious one to take, and what's worse is I cannot figure out how to script it: you've seen that what I've tried in the script above did not work. Meaning I have to do the song and dance of going Import > Extra Cells then navigate to the relevant file, press ok and then press import again to be able to preview a design.

klayoutmatthias, 2021-09-12:
@donn Many thanks for this nice testcase. I always appreciate people taking time for this. It's very helpful for me to understand and debug the problem right away.

In your case, the solution for both issues is adding two settings:

```
    layout_options = tech.load_layout_options
    layout_options.lefdef_config.read_lef_with_def = False
    layout_options.lefdef_config.macro_resolution_mode = 1
    layout_options.lefdef_config.lef_files = ["./merged.lef"]
```

Here is the explanation:

1. The duplicate cell message happens, because by default the DEF reader will first read all LEF files next to the DEF and and afterwards the LEF files you specified explicitly. So it happens that "merged.lef" is read twice. I admit the logic should be a bit smarter and check if a file has been read already. But aside of this, if you give a list of LEF files, you're essentially saying "these are the ones I need", so you can safely turn the default LEF collection off. That is what "read_lef_with_def = False" will do.

2. As the LEF file includes a "FOREIGN" statement, KLayout assumes the layouts will be read from a separate (GDS) file. This is the default behaviour, but you can configure the reader to use the LEF content also if FOREIGN is given. This is what "macro_resolution_mode = 1" will do. If you want the macro cells to be read from a GDS file, consider using "macro_layout_files" to specify the paths to the GDS files to search for the corresponding cells.

Kind regards,

Matthias

donn, 2021-09-12:
Thank you so much for your help! That does get my scripts working. :)

closed 2021-09-12 by donn