KLayout/klayout #909: DRC crash

mguthaus, 2021-09-20:
I am trying to run DRC with the [MOSIS SCMOS DRC deck ](https://github.com/laurentc2/MOSIS_SCMOS_for_KLayout) but I get the following crash. My command line and design are included: klayout -b -r drc_SCMOS.lydrc -rd input=pinv.gds 

> ERROR: Signal number: 11
> Address: 0x30
> Program Version: KLayout 0.27.1 (2021-06-01 r5ccd9990)
> 
> Backtrace:
> /usr/local/klayout/libklayout_lay.so.0 +0x2cf537 lay::enable_signal_handler_gui(bool) [??:?]
> /lib/x86_64-linux-gnu/libc.so.6 +0x46210 killpg [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0x5d8b7c db::Cell::begin_child_cells() const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0x5dc6bd db::Cell::collect_called_cells(std::set<unsigned int, std::less<unsigned int>, std::allocator<unsigned int> >&, int) const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0x110bff4 db::OriginalLayerRegion::count() const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0x10fb041 db::AsIfFlatRegion::add(db::Region const&) const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0xeec37c db::Region::operator+(db::Region const&) const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0xf0a7db void std::vector<db::Region, std::allocator<db::Region> >::emplace_back<db::Region>(db::Region&&) [??:?]
> /usr/local/klayout/libklayout_gsi.so.0 +0x63438 gsi::VariantUserClassImpl::execute_gsi(tl::ExpressionParserContext const&, tl::Variant&, tl::Variant&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<tl::Variant, std::allocator<tl::Variant> > const&) const [??:?]
> /usr/local/klayout/libklayout_gsi.so.0 +0x61d96 gsi::VariantUserClassImpl::execute(tl::ExpressionParserContext const&, tl::Variant&, tl::Variant&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<tl::Variant, std::allocator<tl::Variant> > const&) const [??:?]
> /usr/local/klayout/libklayout_tl.so.0 +0x6d2d6 tl::MethodExpressionNode::execute(tl::EvalTarget&) const [??:?]
> /usr/local/klayout/libklayout_tl.so.0 +0x6bace tl::StaticFunctionExpressionNode::execute(tl::EvalTarget&) const [??:?]
> /usr/local/klayout/libklayout_tl.so.0 +0x5f4d8 tl::Expression::execute() const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0xaf77e9 db::TilingProcessor::execute(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) [??:?]
> /usr/local/klayout/libklayout_tl.so.0 +0xb43bb tl::Worker::run() [??:?]
> /lib/x86_64-linux-gnu/libQt5Core.so.5 +0xbe9d2 QThread::setTerminationEnabled(bool) [??:?]
> /lib/x86_64-linux-gnu/libpthread.so.0 +0x9609 start_thread [??:?]
> /lib/x86_64-linux-gnu/libc.so.6 +0x122293 clone [??:?]

[pinv_example.zip](https://github.com/KLayout/klayout/files/7198776/pinv_example.zip)

mguthaus, 2021-09-20:
This is the most recent klayout results:

> 
> ERROR: Signal number: 11
> Address: 0x30
> Program Version: KLayout 0.27.4 (2021-09-20 r29384bf3f)
> 
> Backtrace:
> /usr/local/klayout/libklayout_lay.so.0 +0x2cf917 lay::signal_handler(int, siginfo_t*, void*) [:?]
> /lib/x86_64-linux-gnu/libc.so.6 +0x46210 killpg [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0x5eb2bc db::Cell::begin_child_cells() const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0x5eeb7d db::Cell::collect_called_cells(std::set<unsigned int, std::less<unsigned int>, std::allocator<unsigned int> >&, int) const [clone .localalias] [dbCell.cc:?]
> /usr/local/klayout/libklayout_db.so.0 +0x1117774 db::OriginalLayerRegion::count() const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0x1105c91 db::AsIfFlatRegion::add(db::Region const&) const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0xef7b3c db::Region::operator+(db::Region const&) const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0xf15f9b gsi::ConstMethod1<db::Region, db::Region, db::Region const&, gsi::arg_default_return_value_preference>::call(void*, gsi::SerialArgs&, gsi::SerialArgs&) const [:?]
> /usr/local/klayout/libklayout_gsi.so.0 +0x63438 gsi::VariantUserClassImpl::execute_gsi(tl::ExpressionParserContext const&, tl::Variant&, tl::Variant&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<tl::Variant, std::allocator<tl::Variant> > const&) const [??:?]
> /usr/local/klayout/libklayout_gsi.so.0 +0x61d96 gsi::VariantUserClassImpl::execute(tl::ExpressionParserContext const&, tl::Variant&, tl::Variant&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<tl::Variant, std::allocator<tl::Variant> > const&) const [??:?]
> /usr/local/klayout/libklayout_tl.so.0 +0x6d356 tl::MethodExpressionNode::execute(tl::EvalTarget&) const [??:?]
> /usr/local/klayout/libklayout_tl.so.0 +0x6bb4e tl::StaticFunctionExpressionNode::execute(tl::EvalTarget&) const [??:?]
> /usr/local/klayout/libklayout_tl.so.0 +0x5f558 tl::Expression::execute() const [??:?]
> /usr/local/klayout/libklayout_db.so.0 +0xb01ac9 db::TilingProcessorWorker::do_perform(db::TilingProcessorTask const*) [:?]
> /usr/local/klayout/libklayout_tl.so.0 +0xb443b tl::Worker::run() [??:?]
> /lib/x86_64-linux-gnu/libQt5Core.so.5 +0xbe9d2 QThread::setTerminationEnabled(bool) [??:?]
> /lib/x86_64-linux-gnu/libpthread.so.0 +0x9609 start_thread [??:?]
> /lib/x86_64-linux-gnu/libc.so.6 +0x122293 clone [??:?]

klayoutmatthias, 2021-09-21:
Hi Matt,

thanks for the test case and the stack trace. I'll try to debug that.

Best regards,

Matthias

klayoutmatthias, 2021-09-21:
Hi Matt,

I could debug and fix the issue. I'm afraid there is no quick workaround, but I'll release the fix with 0.27.4 soon.

Matthias

commit 451f57bb (2021-09-22): Fixed #909 (Crash on DRC)
commit 897bcf1e (2021-09-22): Merge pull request #911 from KLayout/issue-909

Fixed #909 (Crash on DRC)
closed 2021-09-22 by klayoutmatthias