KLayout/klayout #913: DRC report function appends path of source file

mguthaus, 2021-09-22:
The report function in DRC will append the path of the source file so that it gets put in the same directory. I believe that it should be put in the current directory or, if given a relative/absolute path in that location.

klayoutmatthias, 2021-09-23:
Hi Matt,

you're right. But this is somewhat intentional as under Windows, the current directory is often something like "C:\Program Files" or similar. That is not a place I want to write files to :)

However, KLayout prepends the source file's path only if the given file is not an absolute path. In fact, it will resolve relative paths relative to the source file. This means if you use "/home/yourself/myreports/x.lyrdb" for example, it will go there.

If you don't want to put in absolute paths literally, you can use Ruby methods - e.g.

```
report(ENV("HOME") + "/myreports/x.lyrdb")
```

or better

```
report(File.join(ENV("HOME"), "myreports", "x.lyrdb")
```

If you want the report file to go into your current directory, you can use:

```
report(File.join(Dir.pwd, "report.lyrdb")
```

Best regards,

Matthias

mguthaus, 2021-09-24:
Ok, thanks! That seems reasonable. I didn't know that about Windows...

closed 2021-09-24 by mguthaus