KLayout/klayout #914: Improvement: Make "tap" function available in reader mode

3e4, 2021-09-23:
Tap function shows list of layers for polygons below pointer. This function is only available in `klayout -e`. (0.27.3)

Function should be also available in reader mode as it does not alter the layout and is a useful investigation function.

klayoutmatthias, 2021-09-23:
Hi @3e4,

I employed the DeLorean time machine approach from "Back to the future" (https://www.popularmechanics.com/culture/movies/a17862/back-to-the-future-time-machines-schematics/) to implement the solution **before** this ticket was issued: https://github.com/KLayout/klayout/commit/98e20cf2f1a9e6306ca589ce7a285d3f3ac5bb74

I'm only able to cover 8 hours as of now, but I'm working on a larger time leap within this retrograde bug fix approach - specifically before the previous release.

Matthias

klayoutmatthias, 2021-09-26:
Released with 0.27.4

closed 2021-09-26 by klayoutmatthias