KLayout/klayout #929: LVS: multi layer resistor connected through via

anadb, 2021-11-02:
Hi,

I would like to calculate resistance of structure depicted below. This is kind of chain of plates connected through via.
I didn't find out how to do in KLayout since there is only an example with one layer. I tried to make a connectin but it didn't help.

[image]

klayoutmatthias, 2021-11-05:
This is a question, not a bug, right?

I'd like to advertise the forum (https://www.klayout.de/forum/) for support topics.

In your case, the question for me is what is the variable part of the resistor. If that is just the center upper metal part, then you will need some marker or label to identify that part. With this you can use the standard resistor model.

Matthias

anadb, 2021-11-07:
Thank you for your answer, yes this is a question.

I registered on klayout forum but my mail was varified very recently. So I can transfer this question to the forum.
Anyway I need more details

closed 2021-11-07 by anadb