KLayout/klayout #930: Error reading DRC marker file

d-m-bailey, 2021-11-05:
klayout on mac0S won't read the DRC report for via spacing errors. 
version 0.26.10
macOS BigSur
Here's the error message `XML parser error: Expected '/' here: |(1810.12, .. in line 490, column 97`
```
   <category>'via4.2'</category>
   <cell>user_project_wrapper</cell>
   <visited>false</visited>
   <multiplicity>1</multiplicity>
   <image/>
   <values>
    <value>edge-pair: (1810.88,2176.28;1810.88,2177.08)|(1810.12,2177.08;1810.12,2176.28)</value>
   </values>
```
It complains about the `|` between the coordinates. When I change it to `/` like below, it works.
```
   <category>'via4.2'</category>
   <cell>user_project_wrapper</cell>
   <visited>false</visited>
   <multiplicity>1</multiplicity>
   <image/>
   <values>
    <value>edge-pair: (1810.88,2176.28;1810.88,2177.08)/(1810.12,2177.08;1810.12,2176.28)</value>
   </values>
```
[klayout_beol_check.xml.gz](https://github.com/KLayout/klayout/files/7489153/klayout_beol_check.xml.gz)

klayoutmatthias, 2021-11-05:
Please see https://github.com/KLayout/klayout/issues/927. That's exactly the same issue. 

Please use a 0.27.x version of KLayout to open these files.

d-m-bailey, 2021-11-06:
Sorry for the duplicate.

closed 2021-11-06 by d-m-bailey

klayoutmatthias, 2021-11-06:
No worries. You could not see it. It was closed already :)