KLayout/klayout #931: DRC crash

lukasc-ubc, 2021-11-08:
Hi @klayoutmatthias 

We have found that KLayout crashes when running DRC checks, on Windows, version 0.27.4, when there is an assignment of a boolean (TRUE).  It works fine on Mac OSX and Linux however, though I get a warning "warning: constant ::TRUE is deprecated"

[image]
[image]

klayoutmatthias, 2021-11-08:
Hi Lukas,

I think it's a matter of the Ruby version. Windows uses Ruby 3.0.1 already and they seem to have disabled TRUE as alias after deprecating them on Ruby 2.x.

In Ruby, the correct ways are "true" and "false" (lower case).

I don't know why the Windows version crashes ... did you get these errors during PCell evaluation? Maybe that is also related to the 3.0.x Ruby version. 

Matthias

klayoutmatthias, 2022-12-06:
Any feedback? Otherwise I'd close this ticket.