KLayout/klayout #942: technology file routing sufix string typo

kareefardi, 2021-11-25:
Hello. In the lefdef reader, the routing suffix string is wrongly set to `special-routing-suffix-string` making special routing options not persist upon quitting klayout. See the line below 
https://github.com/KLayout/klayout/blob/8bd58be5340a8e9293477628e3111a9e33892226/src/plugins/streamers/lefdef/db_plugin/dbLEFDEFPlugin.cc#L385

klayoutmatthias, 2021-12-06:
Thanks for reporting this and sorry for commenting that late.

That's actually an inherited bug that has been there for long and affects the other "routing" properties too. I'm glad to see someone uses this complex UI at all :)

BTW: the canonical way is to provide a layer mapping file. I was actually even requested to remove all the "computed properties" logic in favour of only the layer map file alone. But seeing the feature is used makes me consider to keep it.

Thanks,

Matthias

closed 2021-12-26 by klayoutmatthias

kareefardi, 2021-12-26:
Hey thanks for your feedback and I am sorry that I missed your response. I don't see a reason to object if you chose to remove this feature. It was kinda intuitive to use through graphical interface. I am also aware of the layers map file. I do remember however that I couldn't easily locate an example for it but I eventually found one somewhere. Thank you and keep up the good work