KLayout/klayout #945: how to keep hidden layers hidden

joamatab, 2021-12-14:
Hi Matthias,

How can we store/maintain the state of shown/hidden layers?

When streaming GDS files to a klayout server it's useful to keep the layout zoom region constant using `current_view`

How can we do the same for layers (keep active layers active and hidden layers hidden)?

When you are modifying a particular layer it's nice to be able to hide all the layers except for the one that you are changing

the only way i was able to get this to work is using `reload`, instead of `load_layout`
https://github.com/gdsfactory/gdsfactory/pull/101

and then i ran into the #122 issue

how can we automatically reload without prompt or store the hidden/active layers from `current_view`? 

Let me know what you think
p.s. thanks for keeping improving klayout

klayoutmatthias, 2021-12-14:
What is a "klayout server"?

The #122 suggestion wasn't implemented for the reasons stated there. It's not sufficient to reload a file upon a file change. You also need to know when writing was finished. That is nothing a file system can provide. You'll need a database system with atomic commits for this purpose. I guess this is what you refer to.

A quick solution is to store the layer properties entirely in a layer properties file and retrieve it into the current view after you have loaded the layout.

Matthias

joamatab, 2021-12-15:
thank you Matthias

by `klayout server` i mean using klayout in server mode, as a visualizer. My layouts are fully scripted and streamed to klayout for verification

that's a wonderful suggestion. what are the functions to store and load layer properties from/to the current view?

joamatab, 2021-12-22:
Hi Matthias,

what are the commands in python to store and load the current layout layer properties?

klayoutmatthias, 2021-12-22:
The methods to load and save layer properties files are found in the LayoutView object and are called "load_layer_props" and "save_layer_props". They take a file path argument which gives the .lyp file from where the properties are loaded or to which they are written.

Matthias

joamatab, 2021-12-24:
Awesome, thank you Matthias!

As you suggested i used "load_layer_props" and "save_layer_props" to keep the same layers when reloading a new GDS file

https://github.com/gdsfactory/gdsfactory/pull/110

it worked perfectly.

I wish you a wonderful day
Joaquin

closed 2022-01-30 by klayoutmatthias