KLayout/klayout #949: Spice model cards

mguthaus, 2021-12-16:
Some libraries (e.g. sky130) use a subckt in spice (lines that start with X) for their transistor rather than a direct BSIM model card (lines with M at the start). They do this because of binning or other model parameters.

Is there a way to have the extract tool in LVS use X instead of M? 

This would likely apply to other extracted devices that are not mosfets as well.

mguthaus, 2021-12-16:
Looks like it is hard coded to M:
https://github.com/KLayout/klayout/blob/e1df15a47ffafd206f68fee338e3fcb09fd922bf/src/db/db/dbNetlistSpiceWriter.cc#L161

klayoutmatthias, 2021-12-26:
The default implementation can be overridden using a "Spice writer delegate". The concept is described here: https://www.klayout.de/doc-qt5/manual/lvs_io.html. There is also the reverse: the spice reader can read subcircuits as devices using a "Spice reader delegate".

Matthias

closed 2022-01-30 by klayoutmatthias