KLayout/klayout #950: Spice extract devices

mguthaus, 2021-12-16:
Is there a way to have device extractions only apply to certain sized devices? 

In sky130, transistors inside the areaid.core are "SRAM" transistors. Either one of the following:
special_pfet_latch
special_nfet_latch
special_pfet_pass
special_nfet_pass

The only difference is the size. The latch devices are W=0.21u L=0.15u while anything else is a pass device.

In addition to this, the memory also has some "parasitic" devices. These are devices that are not quite a "full" nmos or pmos in that the gate does not have a source or drain on one side. What will the W and L extraction do in this case? Here is a picture of poly and diffusion:
[image]

mguthaus, 2021-12-17:
Looks like the answer to the second question is that the parasitic device will be ignored:
https://github.com/KLayout/klayout/blob/e1df15a47ffafd206f68fee338e3fcb09fd922bf/src/db/db/dbNetlistDeviceExtractorClasses.cc#L143
In sky130, however, this is modeled as a mos device with degenerate sizes:

> * tap under poly
> X10 GND GND BL1 GND sky130_fd_pr__special_nfet_latch w=0.21u l=0.08u m=1
> X11 GND GND BR1 GND sky130_fd_pr__special_nfet_latch w=0.21u l=0.08u m=1
> 
> * drainOnly PMOS
> X12 Q_bar WL1 Q_bar VDD sky130_fd_pr__special_pfet_pass W=0.07u L=0.15u m=1
> X13 Q WL0 Q VDD sky130_fd_pr__special_pfet_pass W=0.07u L=0.15u m=1
> 
> * drainOnly NMOS
> X14 GND GND BL1 GND sky130_fd_pr__special_nfet_latch w=0.21u l=0.08u m=1
> X15 GND GND BR1 GND sky130_fd_pr__special_nfet_latch w=0.21u l=0.08u m=1

klayoutmatthias, 2021-12-26:
Hi Matt,

The general idea is to use the geometrical manipulation tools to derive specific device recognition layers.

In the first case, you can split the gates into latch-transistor gates and the gates for the other devices like this:

```
gate = poly & active
sd = active - gate

latch_length = 0.21.um
latch_width = 0.15.um

latch_gate_candidates = gate.with_bbox_min(latch_width).with_bbox_max(latch_length)
latch_gates_lt_width = (latch_gate_candidates.edges & sd).width(latch_width).polygons

latch_gates = latch_gate_candidates.not_interacting(latch_gates_lt_width)

latch_gate_candidates.forget  # don't waste memory for temporary layers
latch_gates_lt_width.forget

other_gates = gate - latch_gates
```

In the second case, the standard device extractors will not produce any device, as the topology is not as expected. One solution was to code a new extractor, but an easier one may be to artificially extend the gate ends by some amount.

I assume the parasitic devices have  a certain length and width only (e.g. L=0.15, W=0.07). So it's basically possible to detect this condition and generate a gate extension which generates a full gate. Like this code:

```
gate = poly & active
sd = active - gate

parasitic_length = 0.15.um
parasitic_width = 0.07.um
parasitic_extension = 0.2.um

parasitic_gate_candidates = gate.with_bbox_min(parasitic_width).with_bbox_max(parasitic_length)
long_edges = parasitic_gate_candidates.edges.with_length(parasitic_length)
long_edges_touching_sd = long_edges & sd
parasitic_gate_extension = long_edges_touching_sd.extended_out(parasitic_extension)
parasitic_gates = parasitic_gate_candidates + parasitic_gate_extension

parasitic_gate_candidates.forget  # don't waste memory for temporary layers
long_edges.forget.forget
long_edges_touching_sd.forget
parasitic_gate_extension.forget
```

This assumes, 200nm gate extension are enough to always cover the distance over the active area.

[image]

Best regards,

Matthias

closed 2022-01-30 by klayoutmatthias