KLayout/klayout #952: 2.5d view - Allow boolean layer opertaion

Grandement, 2021-12-20:
I would kindly ask for a new option for the 2.5d view. It would be very good to have an option in the config to allow boolean layer operation. E.g. to define metal slots by a specific layer.

Reference discussion:
https://www.klayout.de/forum/discussion/1993/2-5d-view-with-boolean-layer-operation#latest

klayoutmatthias, 2022-03-07:
Hi have added a feature for this - in fact I have entirely reworked the whole view generation.

The price to pay is incompatibility: now the 2.5d view uses a script to generate the stack. This script is a superset of the DRC language which includes all the high- and low-end features of DRC, including booleans, sizing etc. In the simple form it is similar to - or at least easy to translate from - the basic form used so far.

klayoutmatthias, 2022-04-09:
0.28 will have this new feature.

closed 2022-04-09 by klayoutmatthias